RfidResearchGroup/proxmark3 #2553: hf mf isen --collect_fm11rf08s - AcquireStaticEncryptedNonces: Auth1 error

robshee81, 2024-10-02:
This is what I get:

[usb] pm3 --> hf mf isen --collect_fm11rf08s
[#] AcquireStaticEncryptedNonces: Auth1 error
[
  [
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"]
  ],
  [
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"],
    ["00000000", "00000000"]
  ],
  [
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"],
    ["0000", "0000"]
  ]
]

So it stops me from being able to use the fm11rf08s_recovery.py script, as it fails at this point or run on its own and I don't know why or how to fix it.

hf mf info

[+] Sector 0 key A... FFFFFFFFFFFF
[+] Sector 0 key B... FFFFFFFFFFFF
[+] Sector 1 key A... FFFFFFFFFFFF
[#] failed reading block
[#] failed reading block
[#] failed reading block
[+] Block 0.......... DB 5C BD 4C 76 08 04 00 03 6B 67 BC 76 26 8A 90

[=] --- Fingerprint
[+] Fudan FM11RF08S

[=] --- Magic Tag Information
[=] <n/a>

[=] --- PRNG Information
[+] Prng................. weak
[+] Static enc nonce..... yes

Anyone able to shed some light? I'd love to be able to use the recovery script.

piotrva, 2024-10-11:
Hi, I have similar problem with some Fudan cards:

**Card 1 (Fingerprint: Fudan FM11RF08):**
```
[usb] pm3 --> hf mf info

[=] --- ISO14443-a Information ---------------------
[+]  UID: 86 EA EB 2C
[+] ATQA: 00 04
[+]  SAK: 08 [2]

[=] --- Keys Information
[+] loaded  2 user keys
[+] loaded 61 keys from hardcoded default array
[+] Sector 0 key A... FFFFFFFFFFFF
[+] Sector 0 key B... FFFFFFFFFFFF
[+] Sector 1 key A... FFFFFFFFFFFF
[+] Backdoor key..... A31667A8CEC1
[+] Block 0.......... 86 EA EB 2C AB 08 04 00 02 54 17 11 53 7F DA 1D | ćŕŰ,ź....T..S.┌.

[=] --- Fingerprint
[+] Fudan FM11RF08

[=] --- Magic Tag Information
[=] <n/a>

[=] --- PRNG Information
[+] Prng....... weak

[usb] pm3 --> hf mf isen --collect_fm11rf08s_with_data --key A31667A8CEC1
[#] AcquireStaticEncryptedNonces: Auth1 error
[+] Saved to json file `C:\ProxSpace\pm3/hf-mf-86EAEB2C-nonces_with_data-006.json`
```
[hf-mf-86EAEB2C-nonces_with_data-006.json](https://github.com/user-attachments/files/17347146/hf-mf-86EAEB2C-nonces_with_data-006.json)

**Card 2 (Fingerprint: Fudan based card):**
```
[usb] pm3 --> hf mf info

[=] --- ISO14443-a Information ---------------------
[+]  UID: 6B 6F 4B 5B
[+] ATQA: 00 04
[+]  SAK: 08 [2]

[=] --- Keys Information
[+] loaded  2 user keys
[+] loaded 61 keys from hardcoded default array
[+] Backdoor key..... A31667A8CEC1
[+] Block 0.......... 6B 6F 4B 5B 14 08 04 00 62 63 64 65 66 67 68 69 | koK[....bcdefghi

[=] --- Fingerprint
[+] Fudan based card

[=] --- Magic Tag Information
[=] <n/a>

[=] --- PRNG Information
[+] Prng....... weak

[usb] pm3 --> hf mf isen --collect_fm11rf08s_with_data --key A31667A8CEC1
[#] AcquireStaticEncryptedNonces: Auth1 error
[+] Saved to json file `C:\ProxSpace\pm3/hf-mf-6B6F4B5B-nonces_with_data-006.json`
```
[hf-mf-6B6F4B5B-nonces_with_data-006.json](https://github.com/user-attachments/files/17347162/hf-mf-6B6F4B5B-nonces_with_data-006.json)

For both cards I can manually read data block using the backdoor key and parameter of `-c 4`

**Card 3 (Fingerprint: Fudan FM11RF32 ; Classic 4k):**
```
[usb] pm3 --> hf mf info

[=] --- ISO14443-a Information ---------------------
[+]  UID: 99 CA 33 36
[+] ATQA: 00 02
[+]  SAK: 18 [2]

[=] --- Keys Information
[+] loaded  2 user keys
[+] loaded 61 keys from hardcoded default array
[+] Sector 0 key A... FFFFFFFFFFFF
[+] Sector 0 key B... FFFFFFFFFFFF
[+] Sector 1 key A... FFFFFFFFFFFF
[+] Backdoor key..... 518B3354E760
[+] Block 0.......... 99 CA 33 36 56 18 02 00 46 44 53 37 30 56 30 31 | .╩36V...FDS70V01

[=] --- Fingerprint
[+] Fudan FM11RF32

[=] --- Magic Tag Information
[=] <n/a>

[=] --- PRNG Information
[+] Prng....... weak

[usb] pm3 --> hf mf isen --collect_fm11rf08s_with_data --key 518B3354E760
[+] Saved to json file `C:\ProxSpace\pm3/hf-mf-99CA3336-nonces_with_data.json`
[usb] pm3 --> script run fm11rf08s_recovery.py
[+] executing python C:\ProxSpace\pm3\proxmark3\client\pyscripts/fm11rf08s_recovery.py
[+] args ''
UID: 99CA3336
Getting nonces...
Generating first dump file
Data have been dumped to `hf-mf-99CA3336-dump.bin`
----Step 1:  0 minutes  4 seconds -----------
Loading mfc_default_keys.dic
Running staticnested_1nt & 2x1nt when doable...
Looking for common keys across sectors...
Computing needed time for attack...
----Step 2:  0 minutes 19 seconds -----------
Still about 12 minutes 10 seconds to run...
Brute-forcing keys... Press any key to interrupt

[+] found keys:

[+] -----+-----+--------------+---+--------------+----
[+]  Sec | Blk | key A        |res| key B        |res
[+] -----+-----+--------------+---+--------------+----
[+]  000 | 003 | ------------ | 0 | ------------ | 0
[+]  001 | 007 | ------------ | 0 | ------------ | 0
[+]  002 | 011 | ------------ | 0 | ------------ | 0
[+]  003 | 015 | ------------ | 0 | ------------ | 0
[+]  004 | 019 | ------------ | 0 | ------------ | 0
[+]  005 | 023 | ------------ | 0 | ------------ | 0
[+]  006 | 027 | ------------ | 0 | ------------ | 0
[+]  007 | 031 | ------------ | 0 | ------------ | 0
[+]  008 | 035 | ------------ | 0 | ------------ | 0
[+]  009 | 039 | ------------ | 0 | ------------ | 0
[+]  010 | 043 | ------------ | 0 | ------------ | 0
[+]  011 | 047 | ------------ | 0 | ------------ | 0
[+]  012 | 051 | ------------ | 0 | ------------ | 0
[+]  013 | 055 | ------------ | 0 | ------------ | 0
[+]  014 | 059 | ------------ | 0 | ------------ | 0
[+]  015 | 063 | ------------ | 0 | ------------ | 0
[+]  032 | 131 | ------------ | 0 | ------------ | 0
[+] -----+-----+--------------+---+--------------+----
[+] ( 0:Failed / 1:Success )

[+] Generating binary key file
[+] Found keys have been dumped to `hf-mf-99CA3336-key.bin`
[=]  --[ FFFFFFFFFFFF ]-- has been inserted for unknown keys
[+] Generating final dump file
[+] Data have been dumped to `hf-mf-99CA3336-dump.bin`
----Step 3: 22 minutes 12 seconds -----------
---- TOTAL: 22 minutes 36 seconds -----------

[+] finished fm11rf08s_recovery.py
```
[hf-mf-99CA3336-nonces_with_data-003.json](https://github.com/user-attachments/files/17347343/hf-mf-99CA3336-nonces_with_data-003.json)

doegox, 2024-10-14:
@robshee81 did you try on the latest version? do you still encounter the same issue?
sometimes some distance between the reader and card may help.

doegox, 2024-10-14:
@piotrva the script is only meant to be run on FM11RF08S, not on FM11RF08 or FM11RF32 (or even older Fudan 62..69)

82ghost82, 2024-11-04:
add me to the party.
The recovery script works well and recovers the key but hf mf isen --collect_fm11rf08s is giving auth1 error

```
[usb] pm3 --> hf mf isen --collect_fm11rf08s
[#] AcquireStaticEncryptedNonces: Auth1 error
[+] Saved to json file `C:\ProxSpace\pm3/hf-mf-3232BBF4-nonces.json`
[usb] pm3 --> hf mf info

[=] --- ISO14443-a Information ---------------------
[+]  UID: 32 32 BB F4
[+] ATQA: 00 04
[+]  SAK: 08 [2]

[=] --- Keys Information
[+] loaded  2 user keys
[+] loaded 61 keys from hardcoded default array
[+] Backdoor key..... A396EFA4E24F
[+] Block 0.......... 32 32 BB F4 4F 08 04 00 04 3B 5B 14 F6 EE 7D 90 | 22╗¶O....;[.÷¯}.

[=] --- Fingerprint
[+] Fudan FM11RF08S

[=] --- Magic Tag Information
[=] <n/a>

[=] --- PRNG Information
[+] Prng....... weak
[+] Static enc nonce... yes
```

doegox, 2024-11-04:
> add me to the party. The recovery script works well and recovers the key (nonces with data file contains correct data) but hf mf isen --collect_fm11rf08s is giving auth1 error

That's normal.
Your card doesn't have a default key in first sector.
You can use the backdoor key to collect the nonces:
`hf mf isen --collect_fm11rf08s -c 4 -k A396EFA4E24F`

I now realize this must be the same issue for @robshee81

closed 2024-11-27 by doegox

badboycxcc, 2025-04-07:
## Same error

```
[usb] pm3 --> hf mf isen --collect_fm11rf08s
[#] AcquireStaticEncryptedNonces: Auth1 error
[+] time: 1085 ms
[+] Saved to json file `D:\APP\PM3\rrg_other-20241121-325376d7ee067f74a5f23566b4469493f870e5b8\client\/hf-mf-028A4FDA-nonces-008.json`
```

## Version

```
  [ Support us, shape the future! ]
     Patreon - https://www.patreon.com/iceman1001/

  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev B
    Memory.... 512 KB ( 65% used )

    Client.... Iceman/master/v4.18994-552-g325376d7e 2024-11-21 20:36:26
    Bootrom... Iceman/master/v4.18994-552-g325376d7e-suspect 2024-11-21 20:35:52
    OS........ Iceman/master/v4.18994-552-g325376d7e-suspect 2024-11-21 20:36:00
    Target.... PM3 GENERIC
```

## Shell

### info
```
[usb] pm3 --> hf mf info

[=] --- ISO14443-a Information ---------------------
[+]  UID: 02 8A 4F DA
[+] ATQA: 00 04
[+]  SAK: 08 [2]

[=] --- Keys Information
[+] loaded  2 user keys
[+] loaded 61 hardcoded keys
[+] Sector 0 key A... FFFFFFFFFFFF
[+] Sector 0 key B... FFFFFFFFFFFF
[+] Backdoor key..... A396EFA4E24F
[+] Block 0.... 028A4FDA1D08040004DB7AEF581AC090 | ..z.X...

[=] --- Fingerprint
[+] Fudan FM11RF08S

[=] --- Magic Tag Information
[=] <n/a>

[=] --- PRNG Information
[+] Prng....... weak
[+] Static enc nonce... yes
```

### isen

```
[usb] pm3 --> hf mf isen

[=] --- ISO14443-a Information ---------------------
[+]  UID: 02 8A 4F DA
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[#] select
[#] auth        cmd: 60 00 | uid: 028a4fda | nr: d27e4afd @| nt: 4312093c @idx 53312| par: 0111 ok
[#] auth nested cmd: 60 00 | uid: 028a4fda | nr: a28eab40 @| nt: 0bcd80e1 @idx 39525| par: 1110 ok | ntenc: f4e23304  | parerr: 1000
[#] Nonce distance: 51748 (first nonce <> nested nonce)
[#] auth nested cmd: 60 00 | uid: 028a4fda | nr: da7df9c6 @| nt: 0bcd80e1 @idx 39525| par: 1110 ok | ntenc: f4e23304  | parerr: 1000
[#] Nonce distance:     0
[=] nTenc f4e23304 par {1110}=000x | ks ff2fb3e5 | nT 0bcd80e1 par 0001 | lfsr16 index 39525
[+] Static enc nonce..... yes
```

### collect_fm11rf08s
```
[usb] pm3 --> hf mf isen --collect_fm11rf08s
[#] AcquireStaticEncryptedNonces: Auth1 error
[+] time: 1094 ms
[+] Saved to json file `D:\APP\PM3\rrg_other-20241121-325376d7ee067f74a5f23566b4469493f870e5b8\client\/hf-mf-028A4FDA-nonces-009.json`
[usb] pm3 --> hf mf isen --collect_fm11rf08s_with_data
[#] AcquireStaticEncryptedNonces: Auth1 error
[+] time: 1120 ms
[+] Saved to json file `D:\APP\PM3\rrg_other-20241121-325376d7ee067f74a5f23566b4469493f870e5b8\client\/hf-mf-028A4FDA-nonces_with_data.json`
```