RfidResearchGroup/proxmark3 #2563: SimulateIso14443aInit uses incompatible pointer types in multiple places

archi, 2024-10-07:
The build fails for several standalone modules with a compiler error.

This is on Arch Linux on x86_64.

The AUR package uses the following `make` invocation pattern to build the standalone modules (I've added the `export` for your convenience):

```
# e.g.
export standalone=HF_FOOBAR
make \
    PLATFORM="PM3OTHER" STANDALONE="${standalone}" \
    FWTAG="other-"$(echo ${standalone} | tr '[:upper:]' '[:lower:]') armsrc/install
```

The invocation with context can be seen at https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=proxmark3-iceman-git#n117
(If you're unfamiliar with the Arch Linux PKGBUILD scripts: Think of it as a shell script, and the `build()` function is invoked) 
As far as I can see no external compiler flags are set.

Since I don't have a RDV4, I emptied the `RDV4_STANDALONE_MODES` array in my local copy of the build script, as to not build these modules.
Maybe some problematic `SimulateIso14443aInit` calls are still hidden in there ;-)

## `standalone=DANKARMULTI`:
```
===================================================================
Version info:      Iceman/makepkg/v4.18994-229-g90c8fa03e
Platform name:     Proxmark3 generic target
PLATFORM:          PM3OTHER
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT COMPRESSION
Standalone mode:   DANKARMULTI
===================================================================
[!] Platform definitions changed, cleaning bootrom/armsrc/recovery first...
===================================================================
Version info:      Iceman/makepkg/v4.18994-229-g90c8fa03e
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.15.15 in /usr/lib)
native BT support: Bluez found, enabled
Jansson library:   system library found
Lua library:       system library found
Python3 library:   Python3 v3.12 found, enabled
GD library:        GD v2.3.3 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  cc (GCC) 14.2.1 20240910
===================================================================
PLATFORM Linux
[*] MAKE fpga_compress/install
[-] CC fpga_compress.c
[-] CC ../../common/lz4/lz4hc.c
[-] CC ../../common/lz4/lz4.c
[=] CXX fpga_compress
[*] MAKE armsrc/install
compiler version:  arm-none-eabi-gcc (Arch Repository) 14.1.0
[-] GEN fpga_version_info.c
[-] CC fpga_version_info.c
[-] GEN obj/fpga_all.bit.z
[-] GEN obj/fpga_all.o
[-] CC start.c
[-] CC iso15693.c
[-] CC ../common/iso15693tools.c
[-] CC thinfilm.c
[-] CC lfops.c
[-] CC lfsampling.c
[-] CC pcf7931.c
[-] CC ../common/lfdemod.c
[-] CC lfadc.c
[... 185 lines cut ...]
../armsrc/Standalone/hf_tcprst.c:373:73: error: passing argument 5 of 'SimulateIso14443aInit' from incompatible pointer type [-Wincompatible-pointer-types]
  373 |             if (SimulateIso14443aInit(tagType, flags, data, &responses, &cuid, counters, tearings, &pages) == false) {
      |                                                                         ^~~~~
      |                                                                         |
      |                                                                         uint32_t * {aka long unsigned int *}
./iso14443a.h:144:114: note: expected 'tag_response_info_t **' but argument is of type 'uint32_t *' {aka 'long unsigned int *'}
  144 | bool SimulateIso14443aInit(uint8_t tagType, uint16_t flags, uint8_t *data, uint8_t *iRATs, tag_response_info_t **responses, uint32_t *cuid, uint32_t counters[3], uint8_t tearings[3], uint8_t *pages);
      |                                                                                            ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
../armsrc/Standalone/hf_tcprst.c:373:90: error: passing argument 7 of 'SimulateIso14443aInit' from incompatible pointer type [-Wincompatible-pointer-types]
  373 |             if (SimulateIso14443aInit(tagType, flags, data, &responses, &cuid, counters, tearings, &pages) == false) {
      |                                                                                          ^~~~~~~~
      |                                                                                          |
      |                                                                                          uint8_t * {aka unsigned char *}
./iso14443a.h:144:150: note: expected 'uint32_t *' {aka 'long unsigned int *'} but argument is of type 'uint8_t *' {aka 'unsigned char *'}
  144 | bool SimulateIso14443aInit(uint8_t tagType, uint16_t flags, uint8_t *data, uint8_t *iRATs, tag_response_info_t **responses, uint32_t *cuid, uint32_t counters[3], uint8_t tearings[3], uint8_t *pages);
      |                                                                                                                                             ~~~~~~~~~^~~~~~~~~~~
../armsrc/Standalone/hf_tcprst.c:373:17: error: too few arguments to function 'SimulateIso14443aInit'
  373 |             if (SimulateIso14443aInit(tagType, flags, data, &responses, &cuid, counters, tearings, &pages) == false) {
      |                 ^~~~~~~~~~~~~~~~~~~~~
./iso14443a.h:144:6: note: declared here
  144 | bool SimulateIso14443aInit(uint8_t tagType, uint16_t flags, uint8_t *data, uint8_t *iRATs, tag_response_info_t **responses, uint32_t *cuid, uint32_t counters[3], uint8_t tearings[3], uint8_t *pages);
      |      ^~~~~~~~~~~~~~~~~~~~~
make[1]: *** [../common_arm/Makefile.common:119: obj/hf_tcprst.o] Error 1
make: *** [Makefile:168: armsrc/install] Error 2
```

archi, 2024-10-07:
This should be fixed by https://github.com/RfidResearchGroup/proxmark3/pull/2564

I was just to daft to realize it's the same issue as with my other PR ;-)

btw, build pm3 without the affected modules and successfully autopwned a genuine Mifare Classic card (MFC1C14_x) with hard prng, which wouldn't work with the [aurora release](https://github.com/RfidResearchGroup/proxmark3/releases/tag/v4.18589) <3

iceman1001, 2024-10-07:
If its fixed,   we can close. 

Nice to hear you managed to autpwn a MFC with hard prng :)

closed 2024-10-07 by iceman1001