RfidResearchGroup/proxmark3 #2568: [FreeBSD] make Proxmark3 portable to FreeBSD

rscheff, 2024-10-12:
trying to run make on the proxmax3 repro shows that some incompatible / non-portable variant of make appears to have been used.

Is there any appetite in making the bulild scripts portable to FreeBSD as well?

make: "/root/proxmark3/Makefile.defs" line 16: Invalid line 'ifneq ($(V),1)', expanded to 'ifneq '
make: "/root/proxmark3/Makefile.defs" line 18: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 66: Variable/Value missing from "export"
make: "/root/proxmark3/Makefile.defs" line 71: Invalid line 'ifeq ($(shell $(CC) -v 2>&1 | grep -c "clang version"), 1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 73: Invalid line 'else'
make: "/root/proxmark3/Makefile.defs" line 75: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 77: Invalid line 'ifeq ($(platform),Darwin)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 78: Invalid line '    ifeq ($(shell uname -p),arm64)', expanded to '    ifeq '
make: "/root/proxmark3/Makefile.defs" line 82: Invalid line '        export IPHONEOS_DEPLOYMENT_TARGET=11.0'
make: "/root/proxmark3/Makefile.defs" line 83: Invalid line '    else'
make: "/root/proxmark3/Makefile.defs" line 86: Invalid line '    endif'
make: "/root/proxmark3/Makefile.defs" line 90: Invalid line 'else'
make: "/root/proxmark3/Makefile.defs" line 93: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 95: Invalid line 'ifeq ($(USE_BREW),1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 96: Invalid line '  ifneq ($(strip $(HOMEBREW_PREFIX)),)', expanded to '  ifneq '
make: "/root/proxmark3/Makefile.defs" line 98: Invalid line '  else'
make: "/root/proxmark3/Makefile.defs" line 100: Invalid line '    ifeq ($(strip $(BREW_PREFIX)),)', expanded to '    ifeq '
make: "/root/proxmark3/Makefile.defs" line 102: Invalid line '    endif'
make: "/root/proxmark3/Makefile.defs" line 103: Invalid line '  endif'
make: "/root/proxmark3/Makefile.defs" line 104: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 106: Invalid line 'ifeq ($(USE_MACPORTS),1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 108: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 110: Invalid line 'ifeq ($(DEBUG),1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 114: Invalid line 'else'
make: "/root/proxmark3/Makefile.defs" line 118: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 120: Invalid line 'ifeq ($(DEBUG_ARM),1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 123: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 125: Invalid line 'ifeq ($(SANITIZE),1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 129: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 137: Invalid line 'ifeq ($(shell expr $(CC_VERSION) \>= 10), 1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 138: Invalid line '  ifneq ($(DETECTED_COMPILER), clang)', expanded to '  ifneq '
make: "/root/proxmark3/Makefile.defs" line 140: Invalid line '  endif'
make: "/root/proxmark3/Makefile.defs" line 141: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 142: Invalid line 'ifeq ($(platform),Darwin)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 143: Invalid line '  ifeq ($(shell uname -p),arm64)', expanded to '  ifeq '
make: "/root/proxmark3/Makefile.defs" line 146: Invalid line '  endif'
make: "/root/proxmark3/Makefile.defs" line 151: Invalid line 'else'
make: "/root/proxmark3/Makefile.defs" line 153: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 175: Invalid line 'ifeq ($(GCCEXTRA),1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 177: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 187: Invalid line 'ifeq ($(CLANGEXTRA),1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 189: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 190: Invalid line 'ifeq ($(CLANGEVERYTHING),1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 192: Invalid line 'endif'
make: "/root/proxmark3/Makefile.defs" line 193: Invalid line 'ifeq ($(NOERROR),1)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile.defs" line 195: Invalid line 'endif'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 23: Invalid line 'define KNOWN_STANDALONE_DEFINITIONS'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 25: Invalid line '| STANDALONE      | DESCRIPTION                            |'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 27: Invalid line '| (empty)         | No standalone mode                     |', expanded to '| '
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 28: Invalid line '+----------------------------------------------------------+'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 29: Invalid line '| LF_SKELETON     | standalone mode skeleton               |'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 30: Invalid line '|                 | - iceman                               |'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 31: Invalid line '+----------------------------------------------------------+'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 32: Invalid line '| LF_EM4100EMUL   | Simulate predefined em4100 tags only   |'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 33: Invalid line '|                 |                                        |'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 34: Invalid line '+----------------------------------------------------------+'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 35: Invalid line '| LF_EM4100RSWB   | Read/simulate/brute em4100 tags &      |'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 36: Invalid line '|                 | clone it to T555x tags                 |'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 37: Invalid line '+----------------------------------------------------------+'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 38: Invalid line '| LF_EM4100RSWW   | Read/simulate/validate em4100 tags &   |'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 39: Invalid line '|                 | clone it to T55xx tags, wipe T55xx tags|'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 40: Invalid line '+----------------------------------------------------------+'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 41: Invalid line '| LF_EM4100RWC    | Read/simulate em4100 tags & clone it   |'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 42: Invalid line '|                 | to T555x tags                          |'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 43: Invalid line '+----------------------------------------------------------+'
make: "/root/proxmark3/armsrc/Standalone/Makefile.hal" line 44: Invalid line '| LF_HIDBRUTE     | HID corporate 1000 bruteforce          |'
[... 407 lines cut ...]
make: "/root/proxmark3/Makefile" line 294: Invalid line 'else ifneq ($(wildcard /etc/fedora-release),)', expanded to 'else ifneq '
make: "/root/proxmark3/Makefile" line 296: warning: duplicate script for target "ifneq" ignored
make: "Makefile" line 48: warning: using previous script for "ifneq" defined here
make: "/root/proxmark3/Makefile" line 296: warning: duplicate script for target "(,)" ignored
make: "Makefile" line 71: warning: using previous script for "(,)" defined here
make: "/root/proxmark3/Makefile" line 297: warning: duplicate script for target "ifneq" ignored
make: "Makefile" line 48: warning: using previous script for "ifneq" defined here
make: "/root/proxmark3/Makefile" line 297: warning: duplicate script for target "(,)" ignored
make: "Makefile" line 71: warning: using previous script for "(,)" defined here
make: "/root/proxmark3/Makefile" line 298: Invalid line 'else'
make: "/root/proxmark3/Makefile" line 301: Invalid line 'endif'
make: "/root/proxmark3/Makefile" line 332: Invalid line 'ifeq ($(platform),Darwin)', expanded to 'ifeq '
make: "/root/proxmark3/Makefile" line 334: Invalid line 'else'
make: "/root/proxmark3/Makefile" line 336: Invalid line 'endif'
make: "/root/proxmark3/Makefile" line 337: Invalid line 'ifneq (,$(EDIT))', expanded to 'ifneq '
make: "/root/proxmark3/Makefile" line 339: Invalid line 'endif'
[... cut ...]
make: "/root/proxmark3/Makefile" line 347: warning: duplicate script for target "ifneq" ignored
make: "Makefile" line 48: warning: using previous script for "ifneq" defined here
make: "/root/proxmark3/Makefile" line 347: warning: duplicate script for target "(,)" ignored
make: "Makefile" line 71: warning: using previous script for "(,)" defined here
make: "/root/proxmark3/Makefile" line 348: Invalid line 'else'
make: "/root/proxmark3/Makefile" line 350: Invalid line 'endif'
make: Fatal errors encountered -- cannot continue
make: stopped in /root/proxmark3

iceman1001, 2024-10-12:
Usually it falls back to yourself.     Make it work on FreeBSD and I merge it!

We welcome pull requests!

nvx, 2024-11-03:
Looking at those errors I expect it's because `make` on FreeBSD is BSD make not GNU make. Try installing `gmake` and running `gmake` instead of `make` and see if that works any better.

iceman1001, 2024-11-03:
is it because iOS etc or  Playstation runs on freebsd?

nvx, 2024-11-03:
> is it because iOS etc or Playstation runs on freebsd?

iOS and Mac OS etc all run Darwin which has its origins in a BSD kernel. I think PlayStation does too but I'm less familiar with it.

iceman1001, 2025-02-20:
Closing because no feedback from original poster.

Using `gmake` should solve the issue.

closed 2025-02-20 by iceman1001