RfidResearchGroup/proxmark3 #2569: Communicating with Proxmark3 device failed

drcobra, 2024-10-13:
**Describe the bug**
When running `auto` I get **Communicating with Proxmark3 device failed** after the **Searching for ISO14443-B tag** 
Tested both windows precompiled and Linux latest source compiled binaries.
Firmware uploading was successful.

**To Reproduce**
Steps to reproduce the behavior:
1. proxmark3 /dev/ttyACM0
2. auto
3. While **Searching for ISO14443-B tag**
4. See error **Communicating with Proxmark3 device failed**

**Expected behavior**
Not to die while searching

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11, Ubuntu
 - inside proxmark3 client run the following commands and paste the output here.

```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.18994-250-g8293b8121-suspect 2024-10-12 16:22:50 88c757715
  compiled with............. MinGW-w64 13.2.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Python SWIG support....... absent
  Lua script support........ present ( 5.4.6 )
  Lua SWIG support.......... present

 [ Proxmark3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.18994-250-g8293b8121-suspect 2024-10-12 16:22:14 88c757715
       os: Iceman/master/v4.18994-250-g8293b8121-suspect 2024-10-12 16:22:24 88c757715
  compiled with GCC 12.2.0

 [ FPGA ]
 fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
 fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
 fpga_pm3_felica.ncd image 2s30vq100 2024-02-03 15:12:41
 fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 64% used )
```

```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 41200
[#]   Available memory........ 41200
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | n/a | n/a |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | n/a | n/a |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | n/a | n/a |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 336896
[#]   Transfer Speed PM3 -> Client... 673792 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 32690 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
```

**Additional context**
The device has recently been from here https://www.aliexpress.com/item/1005007456019349.html
Is there a chance this HW revison is not supported? Or am I doing something wrong?

Thank you.

iceman1001, 2024-10-13:
odd, 
I tried w/o tag   -  it worked
I tried 14a  - it worked
I tried 14b - it worked

All on  WSL-1 / Windows 11,

You are on Proxspace,  which is MINGW   (not ubuntu)
Do you have any more information in order to replicate this?

drcobra, 2024-10-15:
Initially I tested it on 2x Windows native and WSL laptops.

Today I tested it Linux different laptop and I got the same issue.
I have attached the log files.
Have you ever used this HW variation before?
https://www.aliexpress.com/item/1005007456019349.html

[log_20241015114708.txt](https://github.com/user-attachments/files/17377654/log_20241015114708.txt)
[log_20241015115018.txt](https://github.com/user-attachments/files/17377655/log_20241015115018.txt)

[log_20241015114708.txt](https://github.com/user-attachments/files/17377641/log_20241015114708.txt)
[log_20241015115018.txt](https://github.com/user-attachments/files/17377642/log_20241015115018.txt)

drcobra, 2024-10-15:
I tried https://github.com/rfidreadermaker/proxmark3-max and that worked. The GUI.exe has a lot of viruses, but the client proxmark3.exe is good and finally works.

Looks an old version of your software? Any chance adding support for this HW version to your branch?

iceman1001, 2024-10-16:
Feel free to contribute! 

Since you are the one with the hardware

closed 2024-10-16 by iceman1001