RfidResearchGroup/proxmark3 #2574: [idea] Validation of syntax and semantic of NDEF messages (Bachelor's Thesis)

schaefflerf, 2024-10-16:
Hi,
I am about to do my bachelor's thesis in computer science and I was wondering if the following feature is good enough for a pull request (as long as I implement it perfectly)

My goal is to enhance NFC-Security by conducting an in-depth threat analysis and developing a real-time verification script for NDEF messages, focusing on minimizing latency and optimizing throughput. This involves:

- Classifying attack vectors such as character injections and overflows.
- Developing and stress-testing a real-time script for verifying NDEF message syntax and semantics.
- Creating heuristics for detecting typical manipulations in NDEF records/messages.
- Conducting real-world tests to measure system stability and robustness, as well as fine-tuning the solution based on test results.
- Implement the script into the Proxmark3 Iceman Fork (and get an approval for the Pull Request)

**Question:**
Does the Proxmark3 Iceman Fork already include any features or initiatives related to this scope, specifically for real-time NDEF message verification and NFC threat analysis? Any guidance or reference to existing work would be greatly appreciated.

Take a look at an excerpt from my expose for details (for those who have read my post up to this point):

> methodology
> 
> In the context of this work, an extended threat analysis is first conducted to investigate security vulnerabilities in the NFC domain as well as specific threats to NDEF messages. This involves the classification of relevant attack vectors, such as character injections and overflows, which serves as the basis for the testing mechanisms to be developed.
> 
> Building on this, a script for real-time verification of the syntax and semantics of NDEF messages is developed, with a particular focus on optimizing latency and throughput. Comprehensive stress tests are conducted to assess the efficiency of the script under high message frequency.
> 
> Another step involves the development and validation of heuristics for detecting typical manipulations, such as unauthorized characters. These heuristics are tested using scenarios with manipulated messages to evaluate both the detection accuracy and the false alarm rate.
> 
> Additionally, tests are conducted under real conditions to assess the system's stability and robustness. Stability metrics, such as system crashes, and interactions with the Proxmark3 are documented.
> 
> Finally, a comprehensive fine-tuning and evaluation of the test results is conducted to identify potential improvements. Based on practical insights and performance data, the solution is adjusted and optimized.

doegox, 2024-11-27:
Hello, yes it sounds interesting :)
Beware the pm3 code will evolve while you'll work on your topic, so it is better to integrate functionalities regularly than forking, working on your side for months then trying to merge back your work on a codebase that evolved in the meantime.
It's easier and clearer for your evaluators to keep your work as much as possible in a script but we are open to improvements in the client itself if it can help writing scripts (e.g. a more machine-readable output than trying to parse native text output of some verbose proxmark client commands)
If you start your project, join the Discord and don't hesitate to discuss issues in the proxmark3-dev chan.

iceman1001, 2025-02-20:
Interesting topic but no updates in four months.

How are things going?