RfidResearchGroup/proxmark3 #2594: Segfault when calling pm3 from Lua (experimental_lib)

doegox, 2024-10-27:
Quick notes not to forget:

From `client/experimental_lib/example_lua`
```
$ ./test.lua
[=] Session log /home/phil/.proxmark3/logs/log_20241027110937.txt
[+] Using UART port /dev/ttyACM0
[+] Communicating with PM3 over USB-CDC
Segmentation fault
```

Problem appeared somewhere when Lua got migrated from 5.2 to 5.4:
[image]

but to test these old commits, one has to add the missing line
`${PM3_ROOT}/client/src/lua_bitlib.c`
in `client/experimental_lib/CMakeLists.txt` as done later with 50d9042a5cd5415dce0c54ad077529f12cbee108

doegox, 2024-10-27:
@jmichelp could you have a look please?

jmichelp, 2024-10-27:
Just had a go at testing. Works for me after a few tweaks due to OSX:
- had to enforce SKIPQT=1 in the build script otherwise compilation fails (probably some missing CFLAGS as the pre-processor complains about c++17 standard being required)
- had to manually symlink `pm3.so` because the build library is `.dylib` on OSX
- had to also symlink `dkjson.lua` to run the second test in the example otherwise the Lua module wasn't found.

No segfault.

Will test on a Linux next.

jmichelp, 2024-10-27:
Same behavior on Linux after an extra `ln -sf ../../lualibs/dkjson.lua .` to run the tests until the end

jmichelp, 2024-10-27:
Considering that the example relies on the system `lua` binary, my current assumption would be that client and libraries were compiled against the bundled lua5.4 library but you're trying to run the script using your system `lua.5.2` interpreter. The bytecode will be different and would result in a segv.

I could confirm this behavior on my Linux machine by running:
```bash
$ sudo apt remove lua5.4
$ sudo apt install lua5.2
$ ./02run_test.sh
[...]
[+] Communicating with PM3 over USB-CDC
./02run_test.sh: line 8: 2957012 Segmentation fault      ./test.lua
```

doegox, 2024-10-27:
😮🤦‍♂️ that makes totally sense, thanks!
I pushed a fix to call lua5.4 in the shebang and added the dkjson symlink

closed 2024-10-27 by doegox