RfidResearchGroup/proxmark3 #2611: Bug with python scripts and 14b commands

82ghost82, 2024-11-03:
**Describe the bug**
It seems that python scripts (and maybe other types) with 14b commands don't play well with the proxmark.
For example, if I run a simple python script to read a block of a 14b tag, the proxmark "crashes" and it's not able to read the block.
Proxmark client/firmware is updated to commit https://github.com/RfidResearchGroup/proxmark3/commit/5057f1a6844a08a353eeeb9a109717a9ae4e9a15
This was tested on proxspace and linux native with python 3.11 and 3.12.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the attached test script (test.txt) and rename it to .py or code your own to read a block of a 14b tag and save it in client\pyscripts\
[test.txt](https://github.com/user-attachments/files/17609895/test.txt)
3. Connect the proxmark to pc via usb (white led only is on)
4. Put a 14b tag on the hf antenna (I'm using srix4k)
5. Run the pm3 client
6. Run command: script run test.py, choose from menu option '0' (Print information) and press enter
7. Proxmark is not able to read block the block (block 6)
8. Red led on proxmark is on

Workaround:
1. Disconnect the proxmark
2. Download the attached test script (test-fix.txt) and rename it to .py or code your own to read a block of a 14b tag (but include before everything else command 'hf 14b info') and save it in client\pyscripts\; script includes 'hf 14b info' on line 57
[test-fix.txt](https://github.com/user-attachments/files/17609896/test-fix.txt)
4. Connect the proxmark to pc via usb (white led only is on)
5. Put a 14b tag on the hf antenna (I'm using srix4k)
6. Run the pm3 client
7. Run command: script run test-fix.py, choose from menu option '0' (Print information) and press enter
8. Proxmark is able to read block the block (it prints SUCCESS)
9. Exit the script with '1' + enter

**Expected behavior**
Run the script with 14b commands without using the workaround

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Proxspace 3.11/Kali linux

```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/HEAD/v4.18994-387-g5057f1a68-suspect 2024-11-03 10:16:07 c119e9e8f
  compiled with............. MinGW-w64 13.2.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present ( 3.11.5 )
  Python SWIG support....... present
  Lua script support........ present ( 5.4.6 )
  Lua SWIG support.......... present

 [ Proxmark3 ]
  firmware.................. PM3 GENERIC
  external flash............ present

 [ ARM ]
  bootrom: Iceman/HEAD/v4.18994-387-g5057f1a68-suspect 2024-11-02 11:15:01 c119e9e8f
       os: Iceman/HEAD/v4.18994-387-g5057f1a68-suspect 2024-11-02 11:15:16 c119e9e8f
  compiled with GCC 13.3.1 20240614

 [ FPGA ]
 fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
 fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
 fpga_pm3_felica.ncd image 2s30vq100 2024-02-03 15:12:41
 fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 70% used )
```

```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 39428
[#]   Available memory........ 39428
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 302
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Device ID............... 85 / 14 (unknown)
[#]   JEDEC Mfr ID / Dev ID... 85 / 6015
[#]   Unique ID (be).......... 0x0C3537304E325041
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | n/a | n/a |
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | n/a | n/a |
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | n/a | n/a |
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 300032
[#]   Transfer Speed PM3 -> Client... 600064 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 6
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 32172 Hz
[#] Installed StandAlone Mode
[#]  HF 14A SNIFF,  a ISO14443a sniffer with storing in flashmem
[#]
[#] To get the trace from flash and display it:
[#] 1. mem spiffs dump -s hf_14asniff.trace -d hf_14asniff.trace
[#] 2. trace load -f hf_14asniff.trace
[#] 3. trace list -t 14a -1
[#] Flash memory dictionary loaded
[#]
```

iceman1001, 2024-11-03:
I run your test script...   it works like it should.   

[image]

82ghost82, 2024-11-03:
Thank you for testing Iceman, I'm not alone, my friend Sandro has the same issue with mac os Sequoia:
[image]

Moreover, we tested that:
1. Connect pm3 to usb
2. put a srix4k on pm3
3. Run client and issue command "hf 14b rdbl -b 6"
4. No output and red led on on proxmark

Everything is good if:
1. Connect pm3 to usb
2. put a srix4k on pm3
3. Run client and issue command "hf 14b info"
4. Run command "hf 14b rdbl -b 6"

iceman1001, 2024-11-03:
Your script doesn't make sense that it fails.

how about you just run read block and see its output?

82ghost82, 2024-11-03:
It's written in my reply above yours:

> Moreover, we tested that:
> 
>     1. Start fresh: disconnect and connect pm3 to usb
>     2. put a srix4k on pm3
>     3. Run client and issue command "hf 14b rdbl -b 6"
>     4. No output and red led on proxmark
> 
> Everything is good if:
>
>     1. Start fresh: disconnect and connect pm3 to usb
>     2. put a srix4k on pm3
>     3. Run client and issue command "hf 14b info"
>     4. Run command "hf 14b rdbl -b 6"

Read also update2 above; I'm quite sure now that it depends on the compilation environment; I bet you compiled your firmware with an older version of arm gcc than 13.3.1?

iceman1001, 2024-11-03:
`arm-none-eabi-gcc-10.3.1` 

So on ARM webpage  https://developer.arm.com/downloads/-/gnu-rm  the latest version of it is 10.3..

so where you get 13.3.1  ?

82ghost82, 2024-11-03:
> so where you get 13.3.1 ?

I downloaded through apt-get install gcc-arm-none-eabi in kali and it downloaded that version.

Your version is from 2021, this should be the latest release:
https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

gentilkiwi, 2024-11-04:
Strangely looks like same as / maybe related to:

> Strange bug here when playing with 14b (latest FW)
> ```
> [usb] pm3 --> hf 14b dump
> [!] ⚠️  timeout while waiting for reply
> [!] ⚠️  no tag found
> [usb] pm3 --> hf 14b dump
> 
> [!] ⚠️  Communicating with Proxmark3 device failed
> ```
> same multiple time after unplug/replug, it works when I made a 14b info (with success) before 14b dump.
> FPGA image loading related ?

https://discordapp.com/channels/738045391416852520/744489989739380776/1301517650349920318

closed 2024-11-04 by iceman1001

gentilkiwi, 2024-11-04:
Fix problems on my side too :) Thank you @nvx !

PR #2617 (2024-11-04): fix ssp_clk sometimes resetting to 0 shortly after being started or reset
this could happen if TC2 was already 0 when it was started or reset resulting in the initial reset not happening until TC0 had overflowed for the first time as the delay to ensure this didn't happen would be missed when TC2 was already 0

the new behaviour results in TIOA0 being toggled when a software trigger of TC0 happens which makes TC2 reset immediately without having to wait for TC0 to overflow

I tested this quite thoroughly with all the edge cases I could think of around calls to `StartCountSspClk` and `ResetSspC [... cut]