RfidResearchGroup/proxmark3 #2612: hf mf isen displaying static enc nonce info for a tag which is only static nonce

82ghost82, 2024-11-03:
**Describe the bug**
hf mf isen is displaying that the tag has static enc nonce instead of static nonce; the tag is a "special one" with nonce distance = 0.

**To Reproduce**
Steps to reproduce the behavior:
1. Be sure to have a tag with static nonce (non encrypted) with nonce distance = 0
2. Inside pm3 client run isen command, for example: hf mf isen -a -k ffffffffffff -n 2 --addread --blk 11
3. last line shows: Static enc nonce..... yes

Note: for a tag with nonce distance 160 (static, not enc) the output of the command is correct

**Expected behavior**
Do not print static enc nonce info for static nonce tags

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Proxspace 3.11

```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/HEAD/v4.18994-387-g5057f1a68-suspect 2024-11-03 10:16:07 c119e9e8f
  compiled with............. MinGW-w64 13.2.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present ( 3.11.5 )
  Python SWIG support....... present
  Lua script support........ present ( 5.4.6 )
  Lua SWIG support.......... present

 [ Proxmark3 ]
  firmware.................. PM3 GENERIC
  external flash............ present

 [ ARM ]
  bootrom: Iceman/HEAD/v4.18994-387-g5057f1a68-suspect 2024-11-02 11:15:01 c119e9e8f
       os: Iceman/HEAD/v4.18994-387-g5057f1a68-suspect 2024-11-02 11:15:16 c119e9e8f
  compiled with GCC 13.3.1 20240614

 [ FPGA ]
 fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
 fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
 fpga_pm3_felica.ncd image 2s30vq100 2024-02-03 15:12:41
 fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 70% used )
```

```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 39428
[#]   Available memory........ 39428
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 302
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Device ID............... 85 / 14 (unknown)
[#]   JEDEC Mfr ID / Dev ID... 85 / 6015
[#]   Unique ID (be).......... 0x0C3537304E325041
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | n/a | n/a |
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | n/a | n/a |
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | n/a | n/a |
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 300032
[#]   Transfer Speed PM3 -> Client... 600064 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 6
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 32172 Hz
[#] Installed StandAlone Mode
[#]  HF 14A SNIFF,  a ISO14443a sniffer with storing in flashmem
[#]
[#] To get the trace from flash and display it:
[#] 1. mem spiffs dump -s hf_14asniff.trace -d hf_14asniff.trace
[#] 2. trace load -f hf_14asniff.trace
[#] 3. trace list -t 14a -1
[#] Flash memory dictionary loaded
[#]
```

doegox, 2024-11-04:
well technically the info from `mf isen` is correct as the nested nonce is static.
It's just that on this tag the first auth nonce is also static.

doegox, 2024-11-05:
I pushed a change to detect when nested nt == first nt, can you test? I don't have such special tag

82ghost82, 2024-11-05:
> I pushed a change to detect when nested nt == first nt, can you test? I don't have such special tag

Thank you, I tested it:
with static nonce distance 0:
last line is: Static nonce.....yes

with static nonce distance 160:
last line is Nonce distance: 160
No line static nonce: yes

with static enc nonce:
last line is: Static enc nonce.....yes

So, fixed, just missing the line Static nonce.....yes for card distance 160?

doegox, 2024-11-05:
ok I changed a bit the logic again.
it can detect and distinguish between static and "superstatic" (same nt even in nested auths)
BUT only if the requested tests allow to deduce it.
I found one card that has also a superstatic nt=00000000, so:
```
[usb] pm3 --> hf mf isen

[=] --- ISO14443-a Information ---------------------
[+]  UID: 04 11 22 33 44 55 66 
[+] ATQA: 00 42
[+]  SAK: 18 [2]
[#] select
[#] auth        cmd: 60 00 | uid: 33445566 | nr: f8ff06a8 @| nt: 00000000 @idx     0| par: 1111 ok 
[#] auth nested cmd: 60 00 | uid: 33445566 | nr: 0040005a @| nt: 00000000 @idx     0| par: 0011 ok | ntenc: ff1f5b36  | parerr: 1010
[#] Nonce distance:     0 (first nonce <> nested nonce)
[#] auth nested cmd: 60 00 | uid: 33445566 | nr: f03f0dbe @| nt: 00000000 @idx     0| par: 0011 ok | ntenc: ff1f5b36  | parerr: 1010
[#] Nonce distance:     0
[+] Static nonce......... yes, even when nested
```
but `hf mf info` does only say "static" because it's not testing nested auths

Ordinary static:
```
[usb] pm3 --> hf mf isen

[=] --- ISO14443-a Information ---------------------
[+]  UID: 00 56 78 BB 
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[#] select
[#] auth        cmd: 60 00 | uid: 005678bb | nr: 8d699345 @| nt: 01200145 @idx  8125| par: 0000 ok 
[#] auth nested cmd: 60 00 | uid: 005678bb | nr: 4d74752c @| nt: 7eef3586 @idx  8285| par: 0001 ok | ntenc: 8198f939  | parerr: 1010
[#] Nonce distance:   160 (first nonce <> nested nonce)
[#] auth nested cmd: 60 00 | uid: 005678bb | nr: ddfdff02 @| nt: 322bc14d @idx  8445| par: 1010 ok | ntenc: cdf4f83a  | parerr: 1011
[#] Nonce distance:   160
```
It does not show static because it has only one sample of the initial nonce.

If we get more first nonces:
```
[usb] pm3 --> hf mf isen --reset

[=] --- ISO14443-a Information ---------------------
[+]  UID: 00 56 78 BB 
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[#] select
[#] auth        cmd: 60 00 | uid: 005678bb | nr: 5de96b31 @| nt: 01200145 @idx  8125| par: 0000 ok 
[#] auth nested cmd: 60 00 | uid: 005678bb | nr: 9df48d58 @| nt: 7eef3586 @idx  8285| par: 0001 ok | ntenc: 8198f939  | parerr: 1010
[#] Nonce distance:   160 (first nonce <> nested nonce)
[#] select
[#] auth        cmd: 60 00 | uid: 005678bb | nr: 05092cea @| nt: 01200145 @idx  8125| par: 0000 ok 
[#] auth nested cmd: 60 00 | uid: 005678bb | nr: 45028e32 @| nt: 7eef3586 @idx  8285| par: 0001 ok | ntenc: 8198f939  | parerr: 1010
[#] Nonce distance:   160 (first nonce <> nested nonce)
[+] Static nonce......... yes
```

82ghost82, 2024-11-05:
Perfect!tested and fixed, I'm closing this!

closed 2024-11-05 by 82ghost82