RfidResearchGroup/proxmark3 #2622: [python] importing requests module may lead to crash

doegox, 2024-11-07:
Just to track it down as it's probably not an easy fix...

```
$ echo import requests > test.py
$ client/proxmark3 -c "script run test; script run test"
[offline|script] pm3 --> script run test
[+] executing python test.py
[+] args ''

[+] finished test

[offline|script] pm3 --> script run test
[+] executing python test.py
[+] args ''
double free or corruption (out)
Aborted
```

doegox, 2024-11-07:
Failing: python3-requests      2.32.3+dfsg-1

now still Trixie docker but within a venv:
```
python3 -m venv venv
source venv/bin/activate
pip install requests
pip list
    ...
    requests           2.32.3
client/proxmark3 -c "script run test; script run test"
```
and there it works fine :/

doegox, 2024-11-12:
No success so far even by running scripts in some Py_NewInterpreter.
The only way it doesn't crash is by not calling Py_Finalize. Which BTW opens other perspectives as it would allow to preserve variables between script calls...

xianglin1998, 2024-12-19:
It is worth mentioning that the stack size of the Alpine image in Docker is only 80-128KB, but PM3 clients will allocate more stack space than this size at startup, leading to crashes.

https://ariadne.space/2021/06/25/understanding-thread-stack-sizes-and-how-alpine-is-different/

So I am always very sensitive to Docker and client crashes.

doegox, 2024-12-19:
It crashes on my host too (Trixie), the Docker example was just to get an easily reproducible setup.

doegox, 2025-04-09:
Similar crash reported by _shuffle2_ with
```
from pyftdi.i2c import I2cController
i2c = I2cController()
i2c.configure('ftdi://ftdi:232h:FTAYHCZO/1', frequency=10_000)
```
and some C232HM-DDHSL-0 cable.

Crash on `i2c.configure` when the script is invoked twice from the cli.

iceman1001, 2026-09-12:
There was some rpi5 stack liberation which was crashing for someone recently.    

Does this issue still occur ?

doegox, 2026-09-12:
it's unrelated with the stack but it doesn't crash anymore, more probably because of upstream changes in Python libs.

closed 2026-09-12 by doegox