RfidResearchGroup/proxmark3 #2633: Bluetooth device "PM3_RDV4.0" never showing up on laptop or mobile

pentwav, 2024-11-13:
Hello,
I am having an issue.

**Describe the bug**
I was able to flash the device and make it work perfectly, using USB connection.
I also installed the blueshark module, as described in the documentation and in the video here: https://www.youtube.com/watch?v=3D2zh0OiOZs

When I enable the battery and BTpower, I have the blue led blinking. However, when I try to connect to bluetooth, I can't find any device named PM3_RDV4.0 or any other device.

**hw version:**
[ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.18994-478-gb4a310992-suspect 2024-11-12 12:03:49 d94a80c30
  compiled with............. MinGW-w64 13.2.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Python SWIG support....... absent
  Lua script support........ present ( 5.4.6 )
  Lua SWIG support.......... present

 [ Proxmark3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: Iceman/master/v4.18994-478-gb4a310992-suspect 2024-11-12 12:03:15 d94a80c30
       os: Iceman/master/v4.18994-486-g340e63ec2-suspect 2024-11-13 20:59:00 d94a80c30
  compiled with GCC 12.2.1 20221205

 [ FPGA ]
 fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
 fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
 fpga_pm3_felica.ncd image 2s30vq100 2024-02-03 15:12:41
 fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 73% used )

**hw status:**
[#] Memory
[#]   BigBuf_size............. 37188
[#]   Available memory........ 37188
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 120
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Memory size............. 2 mbits / 256 kb
[#]   JEDEC Mfr ID / Dev ID... EF / 3012
[#]   Unique ID (be).......... 0x2BB88963922861D5
[#] Smart card module (ISO 7816)
[#]   version................. v4.42 ( ok )
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | n/a | n/a |
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | n/a | n/a |
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | n/a | n/a |
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 114176
[#]   Transfer Speed PM3 -> Client... 228352 bytes/s
[#] Various
[#]   Max stack usage......... 5208 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 75
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31181 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   Mifare.................. 1991 / 2047 keys
[#]   T55x7................... 124 / 1023 keys
[#]   iClass.................. 29 / 511 keys
[#]

**hw tune:**
[=] -------- Reminder ----------------------------
[=] `hw tune` doesn't actively tune your antennas.
[=] It's only informative.
[=] Measuring antenna characteristics...
[/] 10

[=] -------- LF Antenna ----------
[!] LF antenna............ unusable

[=] -------- HF Antenna ----------
[+] 13.56 MHz............. 50.80 V
[+]
[+] Approx. Q factor measurement
[+] Peak voltage.......... 8.9
[+] HF antenna ( ok )

[-] All values are zero. Not showing LF tuning graph

[=] Q factor must be measured without tag on the antenna

**Additional context**
I tried to compile and flash using the 4.10 of this link https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/bt_manual_v10.md and also using precompiled build for Windows from https://proxmarkbuilds.org/.
I was always able to have the "FPC USART for BT add-on... present", but never any bluetooth device showing up.

Thanks !

pentwav, 2024-11-15:
I found the solution, in Windows is set "Bluetooth devices discovery" to advanced. On my Iphone, I never found a trick to see the PM.

closed 2024-11-15 by pentwav

iceman1001, 2024-11-15:
I think iPhone is BLE and the blueshark is BT2.0.    Which means its not supported with iPhone :(