RfidResearchGroup/proxmark3 #2637: [idea] allow configuration of Bluetooth device name in pm3 wrapper script

Laikulo, 2024-11-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I'd like to be able to configure the device name that pm3 searches for, to allow for use with a device that uses a different name (set via the appropriate AT command)

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

A command line arg to pm3 at a minimum, an envvar would also be nice. Possibly also from a per-user config.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Modifying the pm3 script in repo, and stash/stash poping around pulls.

**Additional context**
Add any other context or screenshots about the feature request here.

iceman1001, 2024-11-15:
The pm3 script tries to automatically detect the different available devices connected to the computer across different OS.
Running `./pm3 --list`  to see which device it finds.   If your bluetooth device name for is not detected then the device most likely have a different com port enumeration.

You can try and debug it

Laikulo, 2024-11-15:
Ah, I was meaning the values that are hard coded here:

https://github.com/RfidResearchGroup/proxmark3/blob/master/pm3#L114

I just modified them in my local copy to the name that I set, and all works as expected, it's just a bit hacky...

iceman1001, 2024-11-15:
I see. 
Modifying the script to handle different names,  you should be able to do it.   I am looking forward to see your fix

closed 2025-02-20 by iceman1001