RfidResearchGroup/proxmark3 #2642: Error/bug in hf mf value?!

ikarus23, 2024-11-18:
Hi, did some testing with inc/dec/transfer/restore and I noticed the following.

This is my sector 1 (not 0):
```
[=]    1 |   4 | 09 00 00 00 F6 FF FF FF 09 00 00 00 00 FF 00 FF | ................ 
[=]      |   5 | 14 00 00 00 EB FF FF FF 14 00 00 00 00 FF 00 FF | ................ 
[=]      |   6 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]      |   7 | FF FF FF FF FF FF 7F 06 98 00 FF FF FF FF FF FF | ................
```
It should be possible to increment block nr 5. But
```
[usb] pm3 --> hf mf value --blk 5 -k FFFFFFFFFFFF --inc 10
[=] Value incremented by : 10
[=] Writing block no 5, key type:A - FFFFFFFFFFFF
[#] Nested auth error
[-] ⛔ Update ... : failed
```

Is there something special with EV1 cards (nested auth)? Did I miss something (I know the ACs do not allow for block 4 to be incremented)?

I'm using the latest build of git on Arch Linux.

82ghost82, 2024-11-19:
Hello, I can't reproduce the issue, command is success with both gen2 magic 1k and mifare classic ev1 1k:

GEN2 CUID 1k
```
[usb] pm3 --> hf mf rdsc -s 1 -b -k FFFFFFFFFFFF

[=]   # | sector 01 / 0x01                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   4 | 09 00 00 00 F6 FF FF FF 09 00 00 00 00 FF 00 FF | ................
[=]   5 | 14 00 00 00 EB FF FF FF 14 00 00 00 00 FF 00 FF | ................
[=]   6 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]   7 | 00 00 00 00 00 00 7F 06 98 00 00 00 00 00 00 00 | ................

[usb] pm3 --> hf mf value --blk 5 -k FFFFFFFFFFFF --inc 10
[=] Value incremented by : 10
[=] Writing block no 5, key A - FFFFFFFFFFFF
[+] Update ... : success
[+] Dec ...... : 30
[+] Hex ...... : 0x1E
[usb] pm3 --> hf mf rdsc -s 1 -b -k FFFFFFFFFFFF

[=]   # | sector 01 / 0x01                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   4 | 09 00 00 00 F6 FF FF FF 09 00 00 00 00 FF 00 FF | ................
[=]   5 | 1E 00 00 00 E1 FF FF FF 1E 00 00 00 00 FF 00 FF | ................
[=]   6 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]   7 | 00 00 00 00 00 00 7F 06 98 00 00 00 00 00 00 00 | ................
```

Mifare classic EV1 1k
```
[usb] pm3 --> hf mf info

[=] --- ISO14443-a Information ---------------------
[+]  UID: 60 9B 81 D0
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[=]
[=] --- Tag Signature
[=]  IC signature public key name: NXP MIFARE Classic MFC1C14_x
[=] IC signature public key value: 044F6D3F294DEA5737F0F46FFEE88A356EED95695DD7E0C27A591E6F6F65962BAF
[=]     Elliptic curve parameters: NID_secp128r1
[=]              TAG IC Signature: 58181C8F836DBDFAFE59096EDD767F5EDCD18BFA1EEB580B1E3D82554B6FDC6C
[+]        Signature verification: successful

[=] --- Keys Information
[+] loaded  2 user keys
[+] loaded 61 keys from hardcoded default array
[+] Sector 0 key A... A0A1A2A3A4A5
[+] Sector 1 key A... FFFFFFFFFFFF

[=] --- Fingerprint
[=] <n/a>

[=] --- Magic Tag Information
[=] <n/a>

[=] --- PRNG Information
[+] Prng....... hard

[usb] pm3 --> hf mf rdsc -s 1 -k FFFFFFFFFFFF

[=]   # | sector 01 / 0x01                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   4 | 09 00 00 00 F6 FF FF FF 09 00 00 00 00 FF 00 FF | ................
[=]   5 | 14 00 00 00 EB FF FF FF 14 00 00 00 00 FF 00 FF | ................
[=]   6 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]   7 | 00 00 00 00 00 00 7F 06 98 00 00 00 00 00 00 00 | ................

[usb] pm3 --> hf mf value --blk 5 -k FFFFFFFFFFFF --inc 10
[=] Value incremented by : 10
[=] Writing block no 5, key A - FFFFFFFFFFFF
[+] Update ... : success
[+] Dec ...... : 30
[+] Hex ...... : 0x1E
[usb] pm3 --> hf mf rdsc -s 1 -k FFFFFFFFFFFF

[=]   # | sector 01 / 0x01                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   4 | 09 00 00 00 F6 FF FF FF 09 00 00 00 00 FF 00 FF | ................
[=]   5 | 1E 00 00 00 E1 FF FF FF 1E 00 00 00 00 FF 00 FF | ................
[=]   6 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]   7 | 00 00 00 00 00 00 7F 06 98 00 00 00 00 00 00 00 | ................
```

ikarus23, 2024-11-19:
Thanks for testing. Very strange. What version of PM3 did you use?

ikarus23, 2024-11-19:
Here are my card info and the log from the try to increment.

```
[usb] pm3 --> hf mf info

[=] --- ISO14443-a Information ---------------------
[+]  UID: 04 00 A6 32 
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[=] 
[=] --- Tag Signature
[=]  IC signature public key name: NXP MIFARE Classic MFC1C14_x
[=] IC signature public key value: 044F6D3F294DEA5737F0F46FFEE88A356EED95695DD7E0C27A591E6F6F65962BAF
[=]     Elliptic curve parameters: NID_secp128r1
[=]              TAG IC Signature: A77A9BA11590CA620FE003DB5F6BB9B87F92813CA7CF37FE7C6E55D279CABAE1
[+]        Signature verification: successful

[=] --- Keys Information
[+] loaded  2 user keys
[+] loaded 61 keys from hardcoded default array
[+] Sector 0 key A... FFFFFFFFFFFF
[+] Sector 0 key B... FFFFFFFFFFFF
[+] Sector 1 key A... FFFFFFFFFFFF
[+] Block 0.......... 04 00 A6 32 90 88 04 00 C8 07 00 20 00 00 00 20 | ...2....... ... 

[=] --- Fingerprint
[+] unknown

[=] --- Magic Tag Information
[=] <n/a>

[=] --- PRNG Information
[+] Prng....... hard

[usb] pm3 --> hf mf value --blk 5 -k FFFFFFFFFFFF --inc 10
[=] Value incremented by : 10
[=] Writing block no 5, key type:A - FFFFFFFFFFFF
[#] Nested auth error
[-] ⛔ Update ... : failed
[usb] pm3 --> hf mf list
[+] Recorded activity ( 214 bytes )
[=] start = start of start frame. end = end of frame. src = source of transfer.
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
       2116 |       4484 | Tag |04  00                                                                   |     | 
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
      10564 |      16388 | Tag |04  00  A6  32  90                                                       |     | 
      19712 |      30176 | Rdr |93  70  04  00  A6  32  90  19  3C                                       |  ok | SELECT_UID
      31300 |      34820 | Tag |08  B6  DD                                                               |  ok | 
      37632 |      42400 | Rdr |60  05  58  2C                                                           |  ok | AUTH-A(5)
      47044 |      51780 | Tag |FD  DD  C5  FD                                                           |     | AUTH: nt
      61184 |      70560 | Rdr |12  6F! 63  A3  98! CD! 31! 80                                           |     | AUTH: nr ar (enc)
      71620 |      76292 | Tag |FF! 3B! DB! 50                                                           |     | AUTH: at (enc)
      82944 |      87648 | Rdr |2B! B7! D4! E6!                                                          |     | 
            |            |  *  |                                              key FFFFFFFFFFFF prng HARD |     |
            |            |  *  |C1  05  7F  9A                                                           |  ok | INC(5)
      88772 |      89412 | Tag |00(4)                                                                    |     | 
            |            |  *  |0A                                                                       |     | 
      95360 |     102368 | Rdr |B4! C3! C1  D9  F1  B0!                                                  |     | 
            |            |  *  |0A  00  00  00  AE  8A                                                   |  ok | 
     242688 |     247456 | Rdr |B9  D9! 95! D2!                                                          |     | 
            |            |  *  |60  00  F5  7B                                                           |  ok | AUTH-A(0)
     252100 |     256836 | Tag |E6! 9F! 54  BC                                                           |     | AUTH: nt (enc)
     266240 |     275616 | Rdr |85  F4! A7! 85! 81  81  3C! 6F!                                          |     | AUTH: nr ar (enc)
```

I noticed your output says
```

[=] --- Fingerprint
[=] <n/a>
```
and mine says
```
[=] --- Fingerprint
[+] unknown
```
so it seams we might have different versions

ikarus23, 2024-11-19:
Tried even more cards. Even one very old one (see blow). Same result.
```
[usb] pm3 --> hf mf info

[=] --- ISO14443-a Information ---------------------
[+]  UID: AA 05 9F D1 
[+] ATQA: 00 04
[+]  SAK: 08 [2]

[=] --- Keys Information
[+] loaded  2 user keys
[+] loaded 61 keys from hardcoded default array
[+] Sector 0 key A... FFFFFFFFFFFF
[+] Sector 0 key B... FFFFFFFFFFFF
[+] Sector 1 key A... FFFFFFFFFFFF
[+] Block 0.......... AA 05 9F D1 E1 88 04 00 47 59 55 D1 41 10 36 07 | ........GYU.A.6.

[=] --- Fingerprint
[+] NXP MF1ICS5006

[=] --- Magic Tag Information
[=] <n/a>

[=] --- PRNG Information
[+] Prng....... weak
```

82ghost82, 2024-11-19:
> Thanks for testing. Very strange. What version of PM3 did you use?

Not the latest, not so old..if you want to try..

```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/HEAD/v4.18994-420-g46813e0e5-suspect 2024-11-05 17:19:37 f22b505ee
  compiled with............. MinGW-w64 13.2.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present ( 3.11.5 )
  Python SWIG support....... present
  Lua script support........ present ( 5.4.6 )
  Lua SWIG support.......... present

 [ Proxmark3 ]
  firmware.................. PM3 GENERIC
  external flash............ present

 [ ARM ]
  bootrom: Iceman/HEAD/v4.18994-420-g46813e0e5-suspect 2024-11-05 17:17:12 f22b505ee
       os: Iceman/HEAD/v4.18994-420-g46813e0e5-suspect 2024-11-05 17:18:06 f22b505ee
  compiled with GCC 12.2.0

 [ FPGA ]
 fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
 fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
 fpga_pm3_felica.ncd image 2s30vq100 2024-02-03 15:12:41
 fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 71% used )
```

iceman1001, 2026-09-12:
This has been fix with the convertion to NG frames.   It was based on a missing initialization of cmd array....

closed 2026-09-12 by iceman1001