RfidResearchGroup/proxmark3 #2667: [idea]

LaBestiaEZ, 2024-11-29:
Is there any way to comunicate with proxmark via web?. Im thinking about using WEB SERIAL API in javascript to com with proxmark but i cant understand how to start coding this

xianglin1998, 2024-12-16:
WebAssembly?

xianglin1998, 2024-12-19:
https://github.com/xianglin1998/proxmark3/tree/webassembly

xianglin1998, 2024-12-19:
How to compile?
1. Install [emscripten ] -> https://emscripten.org/docs/getting_started/downloads.html
2. Follow the tutorial to activate your EMSDK environment.
3. `cd` to PM3 client dir, pay attention!!! must to client dir. not project root.
4. exec compile cmd `emmake make SKIPREADLINE=1 SKIPREVENGTEST=1 V=1`
5. waitting some time...
6. After compile success, exec cmd to test `emrun proxmark3.html -- --flush` (Will open browser for [**proxmark3.html**])

xianglin1998, 2024-12-19:
# What's finish?
1. compile to WASM
2. Run by browser and offline mode is work fine.
3. **STDIN & STDOUT & STDERR** redirect to HTML by **xterm.js**

# Unfinished
1. SIMD for hardnested(Now hardnested is disable, so github workflow will be failure cause by HARDNESTED cmd self check.)
2. [whereami.c] for get path on current env.
3. Serial communication, port like WEBSERIAL, (WEBSERIAL-polyfill), WEBUSB
4. FILE R&W related.

# Note
1. Try to consider compatibility issues with serial communication, including both PC and mobile devices.
2. SIMD needs to consider cross platform compatibility, so only SIMD instructions supported by browsers can be used.
3. How can I conveniently manage files that are operated in the browser sandbox?
4. Can I compile WASM as a workflow task and publish it on the GitHub page for PM3 repo?

jkramarz, 2025-05-11:
Combined with ESP32S3 with UART-to-WebSocket bridge, it may form a perfect mobile client.