RfidResearchGroup/proxmark3 #2674: [idea] Decode trace list when sniffing done

Yenniiii, 2024-12-02:
**Is your feature request related to a problem? Please describe.**
We sniff NFC system with proxmark3 and find that some key personal information is leaked; however, interpreting it required additional effort.

**Describe the solution you'd like**
We wish there was a code in client/src/cmd_trace that could decode it and output it.

**Describe alternatives you've considered**
We want to be able to output refined personal information through the command trace decode after the trace_save to cmd_trace.

iceman1001, 2024-12-03:
Which personal information did you see from that trace?

Yenniiii, 2024-12-04:
We could check student number, name, and birth with student ID card. The student number is a unique number for each student in our school.

iceman1001, 2024-12-05:
You will need to provide sample trace,  sample information in order to made a decoder.

mistial-dev, 2026-04-01:
It sounds like these might be custom encoded MFDES.  They can have a secondary application area that contains any random information.

The SIO doesn't have any place to directly store that data, so it's going to likely be something custom that happens to be on the same card.

iceman1001, 2026-04-01:
OP mentioned they would attach results and I thought that would mean source code and trace file.   

Yet to be seen.

Yenniiii, 2026-04-19:
Thank you for your response, and please accept my apologies for the delayed follow-up.

From our analysis, the identifiable information appeared limited, but because the card also includes a payment function, we could not rule out the possibility that the trace files might contain additional personal data.

For that reason, we did not think it would be appropriate to release the raw trace data without clear consent from the data subjects. I had hoped to contribute the dataset in support of this work, but I have not yet been able to secure that consent.

Instead, I am sharing the decoding code used in our experiment at the following link:
https://github.com/right1magic/2J/blob/master/DJ/DJ.c

Using this code, we observed that both student and staff ID cards revealed identifiable fields such as name, ID number, and date of birth at a specific line. In some cases, additional sensitive identifiers also appeared to be observable.