RfidResearchGroup/proxmark3 #2689: Some scripts use hardcoded tools path

CoelacanthusHex, 2024-12-16:
**Describe the bug**

Some scripts use hardcoded tools path, if packager changes INSTALLTOOLSRELPATH when installing the proxmark3 client (Arch Linux packager did it, see https://gitlab.archlinux.org/archlinux/packaging/packages/proxmark3/-/blob/main/PKGBUILD?ref_type=heads ), it will try to find it in the default path, and fail to find it.

https://github.com/RfidResearchGroup/proxmark3/blob/59ae5d22850542892bdde4af0b844543268604e0/client/pyscripts/fm11rf08s_recovery.py#L54-L55

**To Reproduce**

1. Install Proxmark3 client with `INSTALLTOOLSRELPATH=bin` and `PREFIX=/usr` (so tools will in /usr/bin)
2. `script run fm11rf08_full`
3. See error

```
[usb] pm3 --> script run fm11rf08_full
[+] executing python /usr/bin/../share/proxmark3/pyscripts/fm11rf08_full.py
[+] args ''
Cannot find /usr/share/proxmark3/tools/staticnested_1nt, abort!
[!] ⚠  Script terminated by SystemExit -1
[!] ⚠  finished fm11rf08_full with exception
```

**Expected behavior**

The script runs normally.

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
<details>
<summary>Details</summary>
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.19552-suspect 2024-11-22 12:15:19 a3319015f
  compiled with............. GCC 14.2.1 20240910
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present ( 3.12.7 )
  Python SWIG support....... present
  Lua script support........ present ( 5.4.7 )
  Lua SWIG support.......... present

 [ Proxmark3 ]
  device.................... device / fw mismatch
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.19552-suspect 2024-11-22 12:15:19 a3319015f
       os: Iceman/master/v4.19552-suspect 2024-11-22 12:15:19 a3319015f
  compiled with GCC 14.1.0

 [ FPGA ] 
 fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
 fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
 fpga_pm3_felica.ncd image 2s30vq100 2024-02-03 15:12:41
 fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 72% used )
```

</details>
 - hw status
 

<details>
<summary>Details</summary>
```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 39424
[#]   Available memory........ 39424
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 130
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Mfr ID / Dev ID......... EF / 11
[#]   JEDEC Mfr ID / Dev ID... EF / 3012
[#]   Device.................. W25X20BV
[#]   Memory size............. 256 kB (4 pages * 64k)
[#]   Unique ID (be).......... 0x2C2D5063922861D5
[#] Smart card module (ISO 7816)
[#]   version................. v4.42 ( ok )
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | n/a | n/a | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | n/a | n/a | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | n/a | n/a | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 362496
[#]   Transfer Speed PM3 -> Client... 724992 bytes/s
[#] Various
[#]   Max stack usage......... 5224 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 75
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30891 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   Mifare.................. 2042 / 2047 keys
[#]   T55x7................... 124 / 1023 keys
[#]   iClass.................. 29 / 511 keys
[#] 
```

</details>

 - data tune

**Additional context**
Add any other context about the problem here.

hornet-hacker, 2025-01-10:
There isn't a script called `fm11rf08_full.py`.  The script you quoted a piece of isn't even the same one, as that is `fm11rf08s_recovery.py`.  The portion you quoted is part of an `if` statement that checks whether the script is being run in a Git repo environment, or in an installed environment.  The fact that it is failing is very likely a consequence of installing the proxmark software as a package, which is never recommended since changes in this repo can (and will) break things in downstream distributions.

CoelacanthusHex, 2025-01-12:
A possible solution is to define some constants like TOOLS_PATH in the pm3 Python package. So the scripts maintainers needn't determine the location of these paths.

iceman1001, 2025-02-20:
This would be an issue for @doegox

doegox, 2025-02-20:
I pushed a library to handle the finding of tools and dicts for python scripts.
It gets its paths hardcoded when doing a make install.
This should solve this issue.

closed 2025-02-20 by iceman1001