RfidResearchGroup/proxmark3 #2718: Build fail (symbol issue) with   EMV sim POC merge  (CmdSmartRaw)

nneul, 2025-01-14:
This commit:

commit 4a23fb05f0aa93a85f589d899f687a0dd71a45c6
Author: n-hutton <nathan@zilliqa.com>
Date:   Thu Nov 21 19:23:03 2024 +0000

    Cherry pick POC of emv simulation

Results in:

```/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: obj/emvsim.o: in function `EMVsim':
emvsim.c:(.text.EMVsim+0xc3c): undefined reference to `CmdSmartRaw'
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: emvsim.c:(.text.EMVsim+0xc5a): undefined reference to `CmdSmartRaw'
collect2: error: ld returned 1 exit status
```

It now requires `PLATFORM_EXTRAS=SMARTCARD` to link successfully. 

With this config:

```
# If you want to use it, copy this file as Makefile.platform and adjust it to your needs
# Run 'make PLATFORM=' to get an exhaustive list of possible parameters for this file.

#PLATFORM=PM3RDV4
#PLATFORM=PM3GENERIC
# If you want more than one PLATFORM_EXTRAS option, separate them by spaces:
#PLATFORM_EXTRAS=BTADDON
#PLATFORM_EXTRAS=FLASH
#PLATFORM_EXTRAS=SMARTCARD
#PLATFORM_EXTRAS=BTADDON FLASH
#STANDALONE=LF_SAMYRUN

# Uncomment the line below to set the correct LED order on board Proxmark3 Easy
# Only available with PLATFORM=PM3GENERIC
LED_ORDER=PM3EASY

# Uncomment the lines below in order to make a 256KB image
# and comment out the lines above

PLATFORM=PM3GENERIC
PLATFORM_SIZE=256
STANDALONE=
#SKIP_LF=1
SKIP_HITAG=1
SKIP_EM4x50=1
SKIP_EM4x70=1
#SKIP_ZX8211=1
#SKIP_HF=1
#SKIP_ISO15693=1
SKIP_LEGICRF=1
#SKIP_ISO14443b=1
#SKIP_ISO14443a=1
SKIP_ICLASS=1
SKIP_FELICA=1
SKIP_NFCBARCODE=1
#SKIP_HFSNIFF=1
SKIP_HFPLOT=1

# To accelerate repetitive compilations:
# Install package "ccache" -> Debian/Ubuntu: /usr/lib/ccache, Fedora/CentOS/RHEL: /usr/lib64/ccache
# And uncomment the following line
#export PATH := /usr/lib64/ccache:/usr/lib/ccache:${PATH}

# To install with sudo:
INSTALLSUDO=sudo
```

iceman1001, 2025-01-14:
hm... lets move that emvsim inside "with_smartcard"  directive in the armsrc/Makefile

iceman1001, 2025-01-14:
Pull latest and see if this fixes it

nneul, 2025-01-14:
Changes slightly, needs reference to the EMVsim disabled in main app it looks like:

```
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: obj/appmain.o: in function `AppMain':
appmain.c:(.text.AppMain+0x1242): undefined reference to `EMVsim'
collect2: error: ld returned 1 exit status
```

nneul, 2025-01-14:
Adding a WITH_EMVSIM looks like it'd be straightforward enough if you want a patch

iceman1001, 2025-01-14:
And lets add some guards
try pulling now

nneul, 2025-01-14:
That built clean.

closed 2025-01-14 by iceman1001