RfidResearchGroup/proxmark3 #2727: H10301 broken by

rehall, 2025-01-22:
**Describe the bug**

A recent change caused HID 26bit to be read as Indala 27bit, specifically 32f06db2e where a one character change was made, swapping a 0 for the existing 1.  This breaks the ability to read H10301 tags.

[image]

**To Reproduce**
Steps to reproduce the behavior:
   1. Read a 26 bit HID tag using the current Master.

**Expected behavior**
   1. The tag would be read correctly, but as you can see, it was not:

[image]

**Screenshots**

I reverted that tiny change and rebuilt.  Now an H10301 tag is read correctly:

[image]

**Additional context**

I build new versions of both client and firmware regularly on my Android phone, my MacOs laptop, and my Linux workstation.
Same problem exists on all three, and same fix resolves it everywhere.

iceman1001, 2025-01-22:
yes,  not only that format but more.  We are aware of it.

iceman1001, 2025-02-20:
Should be fixed now.    How about you test latest source?

rehall, 2025-02-23:
Yes, it's fixed everywhere.
I've rebuilt from source on phone, laptop, and workstation.