RfidResearchGroup/proxmark3 #2746: HF Features Cause HW Crash

jsutcliff, 2025-02-06:
I'm having some trouble getting pretty much any of the HF features to work on my hardware. I'm hoping there's something I can do to debug this further but I suspect my hardware is defective somehow.

Basically what happens is, if I run most all commands involving "hf", the device's blue D light turns on and the UART goes unresponsive. This looks almost identical to the behavior in #729 (see video as well). Interestingly, `hf tune` does seem to work and respond to tags. There is no iclass tag present during the search below; it seems to always produce that output.

[image]

Are there any steps I can take to investigate why my device seems to be completely locking up?

I've tried several of the release versions and they all do the same thing. Here's my current config:

 - OS: macOS and Ubuntu
 - hw version
[image]
 - hw status
[image]
 - data tune
[image]

iceman1001, 2025-02-06:
Try latest source and see what happens.   I see its a pm3 easy with new features like spiffs  which is a bit untested. 
It could be you got defective hardware but who knows. 

When you compile using clang take notice if there is any warnings.

jsutcliff, 2025-02-06:
Thanks for taking a look. I went ahead and switched to master and gave it a go on Mac and Ubuntu:

<details>
<summary>Mac Compile</summary>

```bash 
===================================================================
Version info:      Iceman/master/v4.19552-327-gf8abfc073
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   FLASH
Included options:  -DLED_ORDER_PM3EASY FLASH LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT COMPRESSION
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/clean
===================================================================
Version info:      Iceman/master/v4.19552-327-gf8abfc073
Client platform:   Darwin
GUI support:       QT not found, disabled
native BT support: Bluez not found, disabled
Jansson library:   system library not found, using local library
Lua library:       system library found
Python3 library:   Python3 v3.13 found, enabled
GD library:        GD not found, disabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  Apple clang version 14.0.0 (clang-1400.0.29.202)
===================================================================
PLATFORM Darwin
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfc_card_only/clean
[*] MAKE mfc_card_reader/clean
[*] MAKE mfd_aes_brute/clean
[*] MAKE cryptorf/clean

===================================================================
Version info:      Iceman/master/v4.19552-327-gf8abfc073
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   FLASH
Included options:  -DLED_ORDER_PM3EASY FLASH LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT COMPRESSION
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
[*] MAKE bootrom/all
[*] MAKE fpga_compress/all
[*] MAKE mfc_card_only/all
[*] MAKE mfc_card_reader/all
[*] MAKE mfd_aes_brute/all
[*] MAKE cryptorf/all
[-] CC nonce2key.c
[-] CC ../../../common/crapto1/crypto1.c
[-] CC ../../../common/crapto1/crapto1.c
[-] CC ../../../common/bucketsort.c
[-] CC nested_util.c
[-] CC staticnested_0nt.c
[-] CC staticnested_1nt.c
[-] CC staticnested_2nt.c
[-] CC staticnested_2x1nt_rf08s_1key.c
[-] CC staticnested_2x1nt_rf08s.c
[-] CC fpga_compress.c
[... 981 lines cut ...]
[*] MAKE mfd_aes_brute/all
[-] CC brute_key.c
[-] CC util_posix.c
[-] CC randoms.c
[=] CXX brute_key
[-] CC mfd_aes_brute.c
[=] CXX mfd_aes_brute
[-] CC mfd_multi_brute.c
[=] CXX mfd_multi_brute
[*] MAKE cryptorf/all
[-] CC cm.c
[-] CC ../../common/cryptorf/cryptolib.c
[-] CC util.c
[=] CXX cm
[-] CC sm.c
[=] CXX sm
[-] CXX sma.cpp
[=] CXX sma
[-] CXX sma_multi.cpp
[=] CXX sma_multi
```

</details>

Got an openssl warning on Mac but nothing stood out to me using Ubuntu. Both show the same behavior with hf commands.

iceman1001, 2025-02-07:
Join the discord server,    there was some discussion a while ago about some pm3 easy came with faulty HF,     
Could be relevant to your case.  Looking at the correct compilation and the wast amount of people using latest source,  I am quite sure it isn't software related.

jsutcliff, 2025-02-10:
Confirmed faulty hardware after some more experimentation

closed 2025-02-10 by jsutcliff

WiredLife, 2025-03-12:
I have the same problem. Did you found a solution to fix this or can i throw my brandnew rdv4 into the trash? My problems started after using it with the blueshark addon and reading sone mifare desfire cards.