RfidResearchGroup/proxmark3 #2767: [idea] Custom card emulation

lyc8503, 2025-03-02:
**Is your feature request related to a problem? Please describe.**
I'm working on simulating a non-standard NFC card, none of the current `sim` commands work.

**Describe the solution you'd like**
Add a custom simulation mode, when enabled, PM3 forwards all data from reader directly to a lua/python script running on the host PC, and the script decides how to respond.

This apporach can be quite flexible and useful, and it avoid everyone forking, writing C code and compiling firmware on their own when reverse engineering/testing some new NFC cards.

**Additional context**
This feature might be similar to [#628](https://github.com/RfidResearchGroup/proxmark3/issues/628).

iceman1001, 2025-03-02:
Simulation usually involves very tight timings which must uphold. 
This is why you see the "short cuts"  or assumptions NFC relay software does. 
Once you circumvent the anti-collision phase and for instance go from  ISO144433a-3 layer to ISO14443a-4 layer you now have more flexible timings.      
There are no "generic" approach but very custom ones for each application and target technology. 

With that said don't let this stop you and try to prove me wrong by making a working implementation.  
I am looking forward to your pull requests.

lyc8503, 2025-03-04:
Thanks for the explanation, it's true that I didn't consider that the timing could cause problems . 

To be precise, this card I want to simulate follows ISO 7816-4 APDU and ISO 14443-4, in fact what I want to do is to forward the APDUs sent by the reader to the host and use a script on the host to decide how to respond to the reader. I don't know if common readers can accept a certain delay without problems in this case, and if this is a good idea.

The current scripting system uses lua, and I don't know how to write lua, so maybe I'll just have to write my custom logic in C in the firmware first.

closed 2025-03-05 by lyc8503