RfidResearchGroup/proxmark3 #2768: Potentially Incorrect DesFire EV1 Auth Mode Info?

zhiyan114, 2025-03-03:
**Describe the bug**
There seems to be a discrepancy between proxmark3 and NXP TagInfo when it comes to `PICC level auth commands`. When using `mfdes info`, it seems like `PICC level auth commands` is indicating the PICC key as 2TDEA with AES auth being disabled while the official NXP TagInfo is showing that the key is using AES.

I believe this will also cause the `chk` command to test the wrong auth mode since using that command is indicating that it's only testing `2TDEA`?

**To Reproduce**
1. Check the card using NXP TagInfo to make sure the key uses AES
2. Run `hf mfdes info`
3. Run `rf mfdes chk`

**Expected behavior**
PICC Key Mode should match what's on the NXP's TagInfo tool

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 23H2
```
MCU....... AT91SAM7S512 Rev A
Client.... Iceman/master/v4.18994-556-gb1c604c4c 2024-11-22 12:11:28
Bootrom... Iceman/master/v4.18589-402-g16b56f499-suspect 2024-09-10 11:32:01
OS........ Iceman/master/v4.18994-556-gb1c604c4c-suspect 2024-11-22 12:11:02
Target.... RDV4
```

iceman1001, 2025-03-03:
Feel free to find which tools is correct and if the proxmark3 is doing it wrong I am looking forward to your pull request

zhiyan114, 2025-03-03:
Looks like proxmark is the accurate one here. Though, the key type will report `2TDEA` regardless the auth type.

It also looks like there's specific key pattern that must be used for the TagInfo tool to report "Aes Key", I guess that's just some weird implementation from DesFire

closed 2025-03-03 by zhiyan114