RfidResearchGroup/proxmark3 #2769: MIFARE Classic Anti-Copy Detection

stoutouloutou, 2025-03-04:
Hello,
First of all, thanks a lot for all the great work done here, I learned a lot thanks to this software !

I would find it great if the mifare classic scripts could display a warning if detects a "anti-copy" feature while reading a badge / a dump.
The problem and this feature is well described here, seems already implemented for the flipper zero:

[https://github.com/flipperdevices/flipperzero-firmware/issues/1345](https://github.com/flipperdevices/flipperzero-firmware/issues/1345)

Does adding this feature seems relevant ?
(the main goal would be to prevent users from using copies which would lead to a deactivation of their badge, and to learn about anti copy systems... )

Thanks again.

zhiyan114, 2025-03-06:
There's no way for PM3 to detect the suppose "anti-copy" because the said detection is simply by updating the card content per reader scan (and PM3 is just a tool to crack, read, and write the card).

One approach to "detect" it is by dumping the card content then scan it with the reader and dump it again. If the two dumped content are different then there's potentially "anti-copy" implemented.

Eltrick, 2025-03-14:
How a system implements "anti-copy" features is entirely dependent on the system in question, there is no real standard way of implementing such a feature, so there's no real way of implementing such a feature in the proxmark either.

Eltrick, 2025-03-14:
I thought they were talking about anti-cloning features in general and only gave that issue as an example

In either case, I feel like one of the better ways to detect things like this is to warn the user when the reader does a write block command when you try to simulate the dump with the proxmark. In theory this should* be easier to implement since it would basically use the same code as `trace list`'s annotation feature.

Not too sure in practice however.

closed 2026-09-22 by iceman1001