RfidResearchGroup/proxmark3 #2770: [idea] Is there a `hf mfu ndefwrite` command?

jiangyi1985, 2025-03-05:
**Is your feature request related to a problem? Please describe.**
I want to write NDEF data to a NTAG 215 tag. I couldn't find a `hf mfu ndefwrite` command.

**Describe the solution you'd like**
I want to write e.g. url data into this tag.
Tag info as follow:

[image]

[image]

**Describe alternatives you've considered**
Should I use any other command? Didn't find any in `nfc type2` neither.

iceman1001, 2025-06-08:
as you see there is no current implementation of ndef write.    
The normal way is first to "wipe card"  into  "ndef configuration"   if possible.  Most UL/NTAG already has their OTP set. 
next step is to just write the data. 

How to convert NDEF to data array,  I tend to use a `libndef` or similar,   and pipe to via `xxd`    to create the bytes needed.

There is not many active maintainers so feel free to contribute with the needed changes!

0x6r1an0y, 2026-08-23:
Partial progressed in #3488. Not every function implement, but a good start.

closed 2026-09-10 by iceman1001