RfidResearchGroup/proxmark3 #2771: LF tags not wowrking

PrimaLuceEnrico, 2025-03-05:
On a Proxmark3 RDV4 I installed the latest FW (Release v4.19552).
BT is enabled as I have the blueshark module.

All works fine with HF tags (I've succesfully tested some mifare).

With a tag I suppose is a EM tag i get this result:

```
[usb] pm3 --> lf search

[=] Note: False Positives ARE possible
[=] 
[=] Checking for known tags...
[=] 
[=] Odd size,  false positive?
[+] Indala (len 400)  Raw: 800000001fffffffffffffffffffffffffffffffffffffffffffffff

[+] Valid Indala ID found!

[=] Couldn't identify a chipset
```

The Raw response variates a little but is mostly the one above.
I think is not an Indala, anyway I treate it as it would be.
I clone it writing on a t55xx the raw data and read back the same values.
The cloned tag doesen't work on the "original" reader (door access).

If I clone that tag with a simple LF "duplicator" (dupliprox), the cloned tag works at the door.
(It's frustrating that my beloved PM3 can't do what the duplicator does without difficulty.)

If I read the working cloned tag, I get:

```
[usb] pm3 --> lf search

[=] Note: False Positives ARE possible
[=] 
[=] Checking for known tags...
[=] 
[-] ⛔ No known 125/134 kHz tags found!
[+] Chipset detection: T55xx
[?] Hint: try `lf t55xx` commands
```

I succesfully read any kind of EM tag with this PM3 in the past, I'm pretty sure I read EM tag working on the same system (and I remember they were EM). The same happens with a different RDV3 pm3. Nothing changes between BT and USB connection.

I'm supposing something is broken on the LF code.

zhiyan114, 2025-03-06:
Regarding the "EM Tag", try position it differently and see if you can get it to read (i.e. maybe a few CM above the PM3's antenna).

For the "cloned" tag, the output indicates that no valid tag data is written to it, but it is working since it's detecting the tag.

PrimaLuceEnrico, 2025-03-06:
I tried a lot of positions, with "lf tune" at minimum or maximum value, with a CD cover inserted between the tag and the reader. 
If I go far away from the reader, the raw data is variable (= it reads noise)

The "cloned" tag opens the door because it contains valid data, otherwise the door would remain closed (as happens with the tag copied with raw indala data on the pm3)! It is an access control system, not a "tag present in front of the reader" system!

Enrico Vazzoler
Chief Software Projects Officer
PrimaLuceLab S.p.A.
----------------------------------------------------------
Via Torricelli 9
33080 Porcia (PN)

Mail: ***@***.***
Tel: 0039/0434/1696106
Web: www.primalucelab.com

According to italian law D.L. 196/2003 of June 30th 2003 concerning privacy, this message is for the named person's use only. It may contain confidential, proprietary or legally privileged information. If you receive this message by error, please immediately delete it and all copies of it from your system, destroy any hard copies of it and notify the sender. You must not use, disclose, distribute, print, or copy any part of this message if you are not the intended recipient.

wfire06100, 2025-05-19:
Hello,

I have the same problem with an EM 4x50 tag. It used to work fine, now I have Indala.

iceman1001, 2025-06-08:
pull latest and EM4x50 should be fixed now.

elchipeado21, 2025-09-03:
No puedo detectar en cuerpo un implante de 134.00 khz con proxmark3 easy. alguien que me pueda guiar.

iceman1001, 2025-09-03:
and on a RDV4,   flip the Q switch to 7

elchipeado21, 2025-09-06:
Mr. Iceman ¿Se puede piratear NFC kill desde un implante Verichip en el cuerpo, ¿alguien puede enviar una especie de señal al dispositivo para que no funcione? alguien que me pueda guiar.Please

iceman1001, 2026-09-01:
Try pulling latest source,  compile with BTADDON and flash/test again

closed 2026-09-01 by iceman1001