RfidResearchGroup/proxmark3 #2773: encrypted comm mode breaks desfire's clear value operation

zhiyan114, 2025-03-06:
## Context
When setting ValueFileType's communication mode to `encrypt`, the command `hf mfdes value --op clear -m encrypt [... other crypto flags]`, the execution returns the following result:
```
[!] Wrong communication mode. Check settings. command: f5
[=] Key num: 0 Key algo: aes Key[16]: [AES Key]
[=] Secure channel: ev1 Command set: niso Communication mode: encrypt
[=] Session key [16]: [Session Key]
[=]     IV [16]: [IV]
[!] CRC32 error.
[!!] Desfire GetFileSettings command returns wrong data
```
When changing the file communication mode outside of `encrypt`, the command works as intended. I've used DesFire EV1 for the test. Cannot confirm if EV2/EV3 will have this issues.

## Commands to Replicate The Issue
Here's all the command that's been executed leading up to the error.

```
hf mfdes formatpicc
hf mfdes createapp --numkeys 1 --aid 000001 --ks1 09 --dstalgo AES --no-auth
hf mfdes createvaluefile --aid 000001 --fid 01 --amode encrypt --rawrights 0000 --lower 80000000 --upper 7FFFFFFF -t aes
hf mfdes value --aid 000001 --fid 01 -t aes -m encrypt --op clear
```

## Firmware Versions

```
MCU....... AT91SAM7S512 Rev A
Memory.... 512 KB ( 73% used )

Client.... Iceman/master/v4.18994-556-gb1c604c4c 2024-11-22 12:11:28
Bootrom... Iceman/master/v4.18589-402-g16b56f499-suspect 2024-09-10 11:32:01
OS........ Iceman/master/v4.18994-556-gb1c604c4c-suspect 2024-11-22 12:11:02
Target.... RDV4
```

zhiyan114, 2025-09-03:
Yes, the issue still persist. Tested with EV3 card this time though.

Current Build ID:
```
Client.... Iceman/master/v4.20469-233-gd3627920c 2025-09-03 18:16:25
Bootrom... Iceman/master/v4.20469-233-gd3627920c-suspect 2025-09-03 18:15:48 2689d7032
OS........ Iceman/master/v4.20469-233-gd3627920c-suspect 2025-09-03 18:15:57 2689d7032
```

merlokk, 2025-09-21:
clear its not a real command. some errors can occur in some modes. it deeply depends on card/modes and etc.
safe mode to clear - just get and then debit
``` hf mfdes value --aid 000001 --fid 01 -o get -t aes -av -c nat -m enc ```
``` hf mfdes value --aid 000001 --fid 01 -o debit -d 00000005 -t aes -av -c nat -m enc ```
needs to be aware of the min values and etc, that tries to authomate this clear command via real Desfire GetFileSettings command

merlokk, 2025-09-21:
[image]

PR #2987 (2025-09-21): fixed clear command logic. will works with some default settings. 
for non-standard settings, use credit/debit instead of clear command.
[image]

fixes #2773