RfidResearchGroup/proxmark3 #2774: `hf mf gload` not restoring from dump file

stuiterveer, 2025-03-10:
**Describe the bug**
When using `hf mf gload`, it seems to ignore the `-f` parameter. If will provide an error saying "Please specify file or emulator memory", just like when omitting both the `-f` and `--emu` parameters.

**To Reproduce**
1. I have my gen4 magic card set up as new mifare classic 1k with 4-byte UID.
2. I have a dump of a card that has the exact same specifications as the above step.
3. Run `hf mf gload --1k --file <filename>`
4. See the error popping up.

Example output along with the help info for `gload`:
```
[usb] pm3 --> hf mf gload

Load magic gen4 gtu card with data from (bin/eml/json) dump file
or from emulator memory.

usage:
    hf mf gload [-hv] [--mini] [--1k] [--2k] [--4k] [-p <hex>] [-f <fn>] [--emu] [--start <dec>] [--end <dec>]
                   

options:
    -h, --help                     This help
    --mini                         MIFARE Classic Mini / S20
    --1k                           MIFARE Classic 1k / S50 (def)
    --2k                           MIFARE Classic/Plus 2k
    --4k                           MIFARE Classic 4k / S70
    -p, --pwd <hex>                password 4bytes
    -v, --verbose                  verbose output
    -f, --file <fn>                Specify a filename for dump file
    --emu                          from emulator memory
    --start <dec>                  index of block to start writing (default 0)
    --end <dec>                    index of block to end writing (default last block)

examples/notes:
    hf mf gload --emu             
    hf mf gload -f hf-mf-01020304.eml
    hf mf gload -p AABBCCDD --4k -v -f hf-mf-01020304-dump.bin
                                  
    Card must be configured beforehand with `script run hf_mf_ultimatecard`.
    Blocks are 16 bytes long.     

[usb] pm3 --> hf mf gload --1k -f test.bin
[!] ⚠️  Please specify file or emulator memory
```

**Expected behavior**
Successfully save a card dump onto a gen4 magic tag.

**Additional context**
All of this is happening at least on the current latest source:
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.19552-411-g9910a5fee-suspect 2025-03-10 13:57:40 03ecae63e
  compiled with............. GCC 11.4.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present ( 3.10.12 )
  Python SWIG support....... present
  Lua script support........ present ( 5.4.7 )
  Lua SWIG support.......... present

 [ Proxmark3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.19552-411-g9910a5fee-suspect 2025-03-10 13:57:20 03ecae63e
       os: Iceman/master/v4.19552-411-g9910a5fee-suspect 2025-03-10 13:57:39 03ecae63e
  compiled with GCC 10.3.1 20210621 (release)

 [ FPGA ] 
 fpga_pm3_hf.ncd image 2s30vq100 10-03-2025 13:51:17
 fpga_pm3_lf.ncd image 2s30vq100 10-03-2025 13:51:17
 fpga_pm3_felica.ncd image 2s30vq100 10-03-2025 13:51:17
 fpga_pm3_hf_15.ncd image 2s30vq100 10-03-2025 13:51:17

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 74% used )
```

iceman1001, 2025-03-10:
Most likely a consequence from the crashing issue.  Try pulling latest and try again.

stuiterveer, 2025-03-10:
Can confirm it works now!

closed 2025-03-10 by stuiterveer