RfidResearchGroup/proxmark3 #2790: OTP page accepts "zeroing" while emulating MFUL EV1

mishamyte, 2025-03-18:
**Describe the bug**
I'm working with a Vizit reader, which has some filter (also called firewalls in English and Chinese terminology) against copies (variations of USCUID-UL tag) and emulators.
The latest filter can't be bypassed by Proxmark3, because of the wrong implementation of OTP page.

So I use the tag with next data in OTP page (hereinafter: OTP):
```
3/0x03 | CC 3B F5 8C | 1 | .;..
```
Based on the [datasheet](https://www.nxp.com/docs/en/data-sheet/MF0ULX1.pdf):
```
The parameter bytes of the WRITE command and the current contents of the OTP bytes
are bit-wise OR’ed. The result is the new OTP byte contents. This process is irreversible
and once a bit is set to logic 1, **it cannot be changed back to logic 0**.
```

So that concept is used by filter. It reads OTP values, then tries to write `00 00 00 00 ` here, reads it again for validation and writes initial values back.
We are interested in next part of the trace.

For Proxmark3 it's:
```
    3674642 |    3684018 | Rdr |A2  03  00  00  00  00  EB  A2                                           |  ok | WRITEBLOCK(3)
    3686086 |    3686662 | Tag |0A(3)                                                                    |     |
    3718014 |    3722718 | Rdr |30  03  99  9A                                                           |  ok | READBLOCK(3)
    3727410 |    3748274 | Tag |00  00  00  00  77  FF  D8  64  E1  CC  B1  63  4B  A4  94  F0  2E  34   |  ok |
    3846968 |    3856280 | Rdr |A2  03  CC  3B  F5  8C  DA  ED                                           |  ok | WRITEBLOCK(3)
    3858540 |    3859116 | Tag |0A(3)                                                                    |     |
```

For original tag it's:
```
   13601520 |   13610896 | Rdr |A2  03  00  00  00  00  EB  A2                                           |  ok | WRITEBLOCK(3)
   13612068 |   13612708 | Tag |00(4)                                                                    |     |
   13643600 |   13648304 | Rdr |30  03  99  9A                                                           |  ok | READBLOCK(3)
   13864880 |   13869648 | Rdr |50  00  57  CD                                                           |  ok | HALT
```

The same behavior as an original tag has is also implemented in Flipper Zero tool.

As we can see, incorrect behavior in Proxmark3 leads that it can't be used against those (and similar readers) and is incorrect.

Potentially it can be the same for Capability Container bytes of NTAGs, but I had no opportunity to check it.

**Expected behavior**
Proxmark3 will NACK to an invalid attempt and won't change OTP bytes values in emu memory.

**Desktop (please complete the following information):**
 - OS: Win10 (19045.5608)
 -  Client.... Iceman/master/v4.19552-465-g7b528a856-dirty 2025-03-19 00:57:41
    Bootrom... Iceman/master/v4.19552-465-g7b528a856-dirty-suspect 2025-03-19 00:58:31
    OS........ Iceman/master/v4.19552-465-g7b528a856-dirty-suspect 2025-03-19 00:58:54
    Target.... RDV4

**Tag dump and traces from PM3, Tag & Flipper Zero are attached:**
[hf-mfu-04C9D0427A1790-dump.json](https://github.com/user-attachments/files/19329803/hf-mfu-04C9D0427A1790-dump.json)
[traces.zip](https://github.com/user-attachments/files/19329807/traces.zip)

iceman1001, 2025-03-19:
That doesn't make sense.

1.  What is the memory dump you are using?  Share the file please.
2.  What is the simulation output ?
3.  Which system is this reader from?

mishamyte, 2025-03-19:
> That doesn't make sense.
> 
> 1. What is the memory dump you are using?  Share the file please.
> 2. What is the simulation output ?
> 3. Which system is this reader from?

Thanks for asking.
1. The file is attached in initial bug, I use the same dump loaded into a emulator memory ([hf-mfu-04C9D0427A1790-dump.json](https://github.com/user-attachments/files/19329803/hf-mfu-04C9D0427A1790-dump.json))
2. Sim output is attached to my last message ([vizit-mfu-proxmark3-after-fix.zip](https://github.com/user-attachments/files/19339059/vizit-mfu-proxmark3-after-fix.zip))
3. Reader is Vizit RD-5F from Vizit ecosystem

iceman1001, 2025-03-19:
The trace is one thing,  I want to see the pm3 client output / command used.

The simulation is acting really strange.  None of which makes any sense to me.

iceman1001, 2025-03-19:
Join the rfid hacking discord and we can more easily chat there.   ( http://iceman.one )
but did you really flash the device with the changes?

commit af933831 (2025-03-19): fix #2790 -  MFU OTP writes shouldnt allow for zeroing.  Bad sanity check implemented,  it should check that every bit doesnt try to zero out.  Will do for now
closed 2025-03-19 by mishamyte