RfidResearchGroup/proxmark3 #2816: [macOS M1] sed: 1: "pyscripts/pm3_resources.py": extra characters at the end of p command

dstasiuk, 2025-04-02:
On macOS, the installation via Homebrew does not complete successfully and returns error
```
==> make install PREFIX=/opt/homebrew/Cellar/proxmark3/4.20142 BREW_PREFIX=/opt/homebrew PLATFORM=PM3GENERIC SKIPQT=1
Last 15 lines from /Users/danylo/Library/Logs/Homebrew/proxmark3/03.make:
[*] MAKE deps/jansson/libjansson.a
make[2]: Nothing to be done for `all'.
[*] MAKE obj/libmbedtls.a
make[2]: Nothing to be done for `all'.
[*] MAKE deps/reveng/libreveng.a
make[2]: Nothing to be done for `all'.
[*] MAKE deps/tinycbor/tinycbor.a
make[2]: Nothing to be done for `all'.
[*] MAKE deps/whereami/libwhereami.a
make[2]: Nothing to be done for `all'.
[@] Installing client to /opt/homebrew/Cellar/proxmark3/4.20142...
# hack ahead: inject installation path into pm3_resources.py
sed: 1: "pyscripts/pm3_resources.py": extra characters at the end of p command
make[1]: *** [install] Error 1
make: *** [client/install] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/rfidresearchgroup/homebrew-proxmark3/issues

```

This is most likely because `sed` fails to parse the substitution command due to the use of syntax not supported by BSD `sed` (the default on macOS). The issue lies in expressions like `\?`, which are not interpreted as quantifiers for optional characters in BSD `sed` when running in standard mode.

dstasiuk, 2025-04-02:
@doegox The issue is related to this [fix](https://github.com/RfidResearchGroup/proxmark3/commit/a8604e3ff95a429c34343c109decb7559431f20c). Can we add the `-E` flag to enable extended syntax, allowing the use of `?` without a backslash for the macOS-specific branch?

doegox, 2025-04-02:
I'm trusting you on that, I can't test.
Like this @dstasiuk ?
```diff
diff --git a/client/Makefile b/client/Makefile
index 87564f960..014211325 100644
--- a/client/Makefile
+++ b/client/Makefile
@@ -890,11 +890,11 @@ ifneq (,$(INSTALLSHARE))
            $(Q)$(INSTALLSUDO) $(MKDIR) $(DESTDIR)$(PREFIX)$(PATHSEP)$(INSTALLSHARERELPATH)
            # hack ahead: inject installation path into pm3_resources.py
     ifeq ($(platform),Darwin)
-           $(Q)sed -i '' 's|^TOOLS_PATH \?= \?None|TOOLS_PATH="$(PREFIX)$(PATHSEP)$(INSTALLTOOLSRELPATH)"|' pyscripts/pm3_resources.py
-           $(Q)sed -i '' 's|^DICTS_PATH \?= \?None|DICTS_PATH="$(PREFIX)$(PATHSEP)$(INSTALLSHARERELPATH)/dictionaries"|' pyscripts/pm3_resources.py
+           $(Q)sed -E -i '' 's|^TOOLS_PATH \?= \?None|TOOLS_PATH="$(PREFIX)$(PATHSEP)$(INSTALLTOOLSRELPATH)"|' pyscripts/pm3_resources.py
+           $(Q)sed -E -i '' 's|^DICTS_PATH \?= \?None|DICTS_PATH="$(PREFIX)$(PATHSEP)$(INSTALLSHARERELPATH)/dictionaries"|' pyscripts/pm3_resources.py
            $(Q)$(INSTALLSUDO) $(CP) $(INSTALLSHARE) $(DESTDIR)$(PREFIX)$(PATHSEP)$(INSTALLSHARERELPATH)
-           $(Q)sed -i '' 's|^TOOLS_PATH \?=.*|TOOLS_PATH = None|' pyscripts/pm3_resources.py
-           $(Q)sed -i '' 's|^DICTS_PATH \?=.*|DICTS_PATH = None|' pyscripts/pm3_resources.py
+           $(Q)sed -E -i '' 's|^TOOLS_PATH \?=.*|TOOLS_PATH = None|' pyscripts/pm3_resources.py
+           $(Q)sed -E -i '' 's|^DICTS_PATH \?=.*|DICTS_PATH = None|' pyscripts/pm3_resources.py
     else
            $(Q)sed -i 's|^TOOLS_PATH \?= \?None|TOOLS_PATH="$(PREFIX)$(PATHSEP)$(INSTALLTOOLSRELPATH)"|' pyscripts/pm3_resources.py
            $(Q)sed -i 's|^DICTS_PATH \?= \?None|DICTS_PATH="$(PREFIX)$(PATHSEP)$(INSTALLSHARERELPATH)/dictionaries"|' pyscripts/pm3_resources.py
```

dstasiuk, 2025-04-03:
@doegox Something's wrong — this issue occurs with the Homebrew installation, but not with the native `make client/install` run. Note: brew dev installation only works from a branch/commit from this repo, not from a fork. 
Could you create a branch with these changes? I’ll be able to run it on my end.

doegox, 2025-04-03:
@dstasiuk  I pushed the proposed change in https://github.com/RfidResearchGroup/proxmark3/tree/osx_test

doegox, 2025-04-03:
please test also that it doesn't break the native make client/install on osx

dstasiuk, 2025-04-03:
@doegox Great, it works both with Brew and natively!

```
>> HOMEBREW_TRAVIS_COMMIT=899e5719577c900ae82b5de6a048cfb789a1c342 HOMEBREW_TRAVIS_BRANCH=osx_test brew install --with-generic --HEAD proxmark3
==> Fetching dependencies for rfidresearchgroup/proxmark3/proxmark3: lua
==> Fetching lua
==> Downloading https://ghcr.io/v2/homebrew/core/lua/manifests/5.4.7
Already downloaded: /Users/danylo/Library/Caches/Homebrew/downloads/d2210560a2e24627c19106d7d878e251812e5f4f990c9fe13887746bfd4d7a21--lua-5.4.7.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/lua/blobs/sha256:925b155912162179d777cde03b60fdd8b5507f0bfc3c7adf2f7a49e1b3b00461
Already downloaded: /Users/danylo/Library/Caches/Homebrew/downloads/1d0e3e66b227f12f82fbe7c27cca3d81db1b929f2d0c0db0b1b71d6ed9867dcb--lua--5.4.7.arm64_sequoia.bottle.tar.gz
==> Fetching rfidresearchgroup/proxmark3/proxmark3
==> Cloning https://github.com/RfidResearchGroup/proxmark3.git
Updating /Users/danylo/Library/Caches/Homebrew/proxmark3--git
==> Checking out branch osx_test
Already on 'osx_test'
Your branch is up to date with 'origin/osx_test'.
HEAD is now at 899e57195 Fix homebrew install
==> Installing proxmark3 from rfidresearchgroup/proxmark3
==> Installing dependencies for rfidresearchgroup/proxmark3/proxmark3: lua
==> Installing rfidresearchgroup/proxmark3/proxmark3 dependency: lua
==> Downloading https://ghcr.io/v2/homebrew/core/lua/manifests/5.4.7
Already downloaded: /Users/danylo/Library/Caches/Homebrew/downloads/d2210560a2e24627c19106d7d878e251812e5f4f990c9fe13887746bfd4d7a21--lua-5.4.7.bottle_manifest.json
==> Pouring lua--5.4.7.arm64_sequoia.bottle.tar.gz
🍺  /opt/homebrew/Cellar/lua/5.4.7: 30 files, 789.5KB
==> Installing rfidresearchgroup/proxmark3/proxmark3 --HEAD --with-generic
==> make clean
==> make all BREW_PREFIX=/opt/homebrew PLATFORM=PM3GENERIC SKIPQT=1
==> make install PREFIX=/opt/homebrew/Cellar/proxmark3/HEAD-899e571 BREW_PREFIX=/opt/homebrew PLATFORM=PM3GENERIC SKIPQT=1
==> Install success!
==> The latest bootloader and firmware binaries are ready and waiting in the current homebrew Cellar within share/firmware.
🍺  /opt/homebrew/Cellar/proxmark3/HEAD-899e571: 788 files, 51.4MB, built in 56 seconds
==> Running `brew cleanup proxmark3`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
```

doegox, 2025-04-03:
thanks for testing! I'll merge it

closed 2025-04-03 by doegox

wesbos, 2025-04-10:
How do I use this fix? The current brew install still suffers from this issue

pgrenaud, 2025-04-22:
> `brew install --HEAD --with-generic proxmark3` for anyone finding this

Just a small precision, adding `--HEAD` _is_ the solution, while adding `--with-generic` is unrelated to this problem. [See this section of the wiki.](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/macOS-Homebrew-Installation-Instructions.md#install-proxmark3-tools)

I encountered this problem myself and installing from HEAD fixed the issue.