RfidResearchGroup/proxmark3 #2835: hf mf simulator cannot read past block 32

tpt-miedwards, 2025-04-30:
This occurs both on a fresh start of pm3 without loading a saved dump, and also with a saved dump loaded.

Symptoms are:
```
[usb] pm3 --> hf mf ekeyprn --1k
[!] ⚠️  error get block 35
[!] ⚠️  error get block 39
[!] ⚠️  error get block 43
[!] ⚠️  error get block 47
[!] ⚠️  error get block 51
[!] ⚠️  error get block 55
[!] ⚠️  error get block 59
[!] ⚠️  error get block 63

[+] -----+-----+--------------+---+--------------+----
[+]  Sec | Blk | key A        |res| key B        |res
[+] -----+-----+--------------+---+--------------+----
[+]  000 | 003 | 000000000000 | 1 | 000000000000 | 1 
[+]  001 | 007 | 000000000000 | 1 | 000000000000 | 1 
[+]  002 | 011 | 000000000000 | 1 | 000000000000 | 1 
[+]  003 | 015 | 000000000000 | 1 | 000000000000 | 1 
[+]  004 | 019 | 000000000000 | 1 | 000000000000 | 1 
[+]  005 | 023 | 000000000000 | 1 | 000000000000 | 1 
[+]  006 | 027 | 000000000000 | 1 | 000000000000 | 1 
[+]  007 | 031 | 000000000000 | 1 | 000000000000 | 1 
[+]  008 | 035 | ------------ | 0 | ------------ | 0 
[+]  009 | 039 | ------------ | 0 | ------------ | 0 
[+]  010 | 043 | ------------ | 0 | ------------ | 0 
[+]  011 | 047 | ------------ | 0 | ------------ | 0 
[+]  012 | 051 | ------------ | 0 | ------------ | 0 
[+]  013 | 055 | ------------ | 0 | ------------ | 0 
[+]  014 | 059 | ------------ | 0 | ------------ | 0 
[+]  015 | 063 | ------------ | 0 | ------------ | 0 
[+] -----+-----+--------------+---+--------------+----
[+] ( 0:Failed / 1:Success )
```

Dumping sector 7 works, as expected:
```
[usb] pm3 --> hf mf egetsc -s 7

[=]   # | sector 07 / 0x07                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]  28 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  29 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  30 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=]  31 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
```

Dumping sector 8, however, fails after block 32:
```
[usb] pm3 --> hf mf egetsc -s 8

[=]   # | sector 08 / 0x08                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]  32 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
```

Interestingly, `hf mf eview` works, both with the default emulator memory and with a dump loaded, but every other sim-related command fails when attempting to read block 33 or higher.

**Desktop (please complete the following information):**
OS: Linux
 ```
[usb] pm3 --> hw version

 [ Proxmark3 ]

 [ Client ]
  Iceman/master/v4.20142-88-g42cdd34e5-suspect 2025-04-30 19:20:54 214a14c48
  Compiler.................. GCC 12.2.0
  Platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  Native BT support......... present
  Python script support..... present ( 3.11.2 )
  Python SWIG support....... present
  Lua script support........ present ( 5.4.7 )
  Lua SWIG support.......... present

 [ Model ]
  Firmware.................. PM3 GENERIC

 [ ARM ]
  Bootrom.... Iceman/master/v4.20142-88-g42cdd34e5-suspect 2025-04-30 19:20:49 214a14c48
  OS......... Iceman/master/v4.20142-88-g42cdd34e5-suspect 2025-04-30 19:20:59 214a14c48
  Compiler... GCC 12.2.1 20221205

 [ FPGA ] 
 fpga_pm3_hf.ncd image 2s30vq100 29-04-2025 16:58:32
 fpga_pm3_lf.ncd image 2s30vq100 29-04-2025 16:58:32
 fpga_pm3_felica.ncd image 2s30vq100 29-04-2025 16:58:32
 fpga_pm3_hf_15.ncd image 2s30vq100 29-04-2025 16:58:32

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 68% used )
```

```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 39976
[#]   Available memory........ 35880
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 29-04-2025 16:58:32
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | n/a | n/a | 
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | n/a | n/a | 
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | n/a | n/a | 
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.............. std    ( follow standard )
[#]   [b] BCC override.................. std    ( follow standard )
[#]   [2] CL2 override.................. std    ( follow standard )
[#]   [3] CL3 override.................. std    ( follow standard )
[#]   [r] RATS override................. std    ( follow standard )
[#]   [m] Magsafe polling............... disabled
[#]   [p] Polling loop annotation....... disabled 00000000000000000000000000000000
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 375296
[#]   Transfer Speed PM3 -> Client... 750592 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 32004 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] 
```

`skipping hw tune - this is irrelevant to this issue`

iceman1001, 2025-05-01:
Interesting,   I pushed a fix ,  pull latest and test

closed 2025-05-01 by iceman1001

tpt-miedwards, 2025-05-02:
Great, that seems to fix it.  Thanks!

commit c847896a (2025-05-01): fix #2835  due to wrong variable used to calculate the size of the emulator memory to download , all get memory from emulator memory failed with a block no larger that 30-ish