RfidResearchGroup/proxmark3 #2842: Changing type of application key on Mifare Desfire does not seem to work

OevreFlataeker, 2025-05-09:
***Things to try before submitting bug report***
read the [troubleshooting guide](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to Iceman repo.

-

**Describe the bug**
A clear and concise description of what the bug is.
Please include text output of the bug happening.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - hw tune

**Additional context**
Add any other context about the problem here.

OevreFlataeker, 2025-05-09:
I am experimenting with a Mifare Desfire EV1 tag and the following proxmark version:

Client.... Iceman/master/v4.20142-91-gb36b61feb 2025-05-04 17:07:38
Bootrom... Iceman/master/v4.20142-57-ga8a1c7734-suspect 2025-04-20 10:55:19 c9
OS........ Iceman/master/v4.20142-57-ga8a1c7734-suspect 2025-04-20 10:55:59 c960d7c

I am able to change the tag's masterkey and create new applications and files on the tag. I can change the AMK. What doesn´t work is changing the type of the AMK from DES to AES for example. The command fails with an error about the command lenght:

As an example I authenticate with key 1, in mode DES and try to change the key 2 from DES to AES and set a new key

```
usb] pm3 --> hf mfdes changekey --aid 111114 --schann ev1 -n 1 -t des -k 0000000000000000 --newkeyno 2 --newalgo aes --newkey 11111111111111111111111111111111  -a -v --cmode encrypt --newver 01
[=] Changing key for AID 111114
[=] auth key 1: des [8] 00 00 00 00 00 00 00 00 
[=] changing key number  0x02 (2)
[=] old key: des [8] 00 00 00 00 00 00 00 00 
[=] new key: aes [16] 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[=] new key version: 0x00
[=] Key num: 1 Key algo: des Key[8]: 00 00 00 00 00 00 00 00 
[=] Secure channel: n/a Command set: niso Communication mode: encrypt
[+] Setting ISODEP -> inactive
[+] >>>> 90 5A 00 00 03 14 11 11 00 
[+] Setting ISODEP -> inactive
[+] Setting ISODEP -> NFC-A
[+] <<<< 91 00 
[=] AID 111114 is selected
[=] Auth: cmd: 0x1a keynum: 0x01
[+] >>>> 90 1A 00 00 01 01 00 
[+] <<<< CF 6B F1 E0 0D B3 91 14 91 AF 
[+] >>>> 90 AF 00 00 10 E3 7F 68 89 66 11 FC 94 2C 42 D2 29 A3 16 4C C1 00 
[+] <<<< B2 CC D4 01 87 06 E5 86 91 00 
[=] Session key : 01 02 03 04 D1 F3 96 66 
[=] Desfire  authenticated
[+] >>>> 90 C4 00 00 21 02 9C FC 3E 51 EE 20 4D 65 53 39 25 D6 06 33 9A 1E 13 FF 08 B0 79 38 EF 7C 72 11 44 31 D5 F4 5E 61 00 
[+] <<<< 91 7E 
[!!] 🚨 APDU(90c4) ERROR: [0x917E] Length of command string invalid
[-] ⛔ Change key ( failed )
[+] Setting ISODEP -> inactive
``` 

I tried with all kind of variations of the command for e.g. secure channel, communication mode, different key, etc. but I always get an error 0x917e back.

Also I noticed that although I specify --newver 01 the commands output still shows "new key version: 0x00" instead of the expected 0x01.

Another example, this time authenticate with key 0 and change key 1, without scannel and encryption type:

```
[usb] pm3 --> hf mfdes changekey --aid 111114 -n 0 -t des -k 1111111111111111 --newkeyno 1 --newalgo aes --newkey 11111111111111111111111111111111  -a -v
[=] Changing key for AID 111114
[=] auth key 0: des [8] 11 11 11 11 11 11 11 11 
[=] changing key number  0x01 (1)
[=] old key: des [8] 00 00 00 00 00 00 00 00 
[=] new key: aes [16] 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[=] new key version: 0x00
[=] Key num: 0 Key algo: des Key[8]: 11 11 11 11 11 11 11 11 
[=] Secure channel: n/a Command set: niso Communication mode: encrypt
[+] Setting ISODEP -> inactive
[+] >>>> 90 5A 00 00 03 14 11 11 00 
[+] Setting ISODEP -> inactive
[+] Setting ISODEP -> NFC-A
[+] <<<< 91 00 
[=] AID 111114 is selected
[=] Auth: cmd: 0x1a keynum: 0x00
[+] >>>> 90 1A 00 00 01 00 00 
[+] <<<< DB 6D F9 8C 52 9C F6 25 91 AF 
[+] >>>> 90 AF 00 00 10 07 31 4F FB 5F 42 17 36 58 CD A9 0E 4D 70 A4 C8 00 
[+] <<<< D5 94 8A EF 5C B6 AF DE 91 00 
[=] Session key : 01 02 03 04 8F 98 76 BF 
[=] Desfire  authenticated
[+] >>>> 90 C4 00 00 21 01 1C F1 C5 65 10 7F 54 F1 08 7E B7 1A 44 8C D2 04 10 95 2D DE A9 29 60 65 0B 1A 98 1C 90 BB 41 A4 00 
[+] <<<< 91 7E 
[!!] 🚨 APDU(90c4) ERROR: [0x917E] Length of command string invalid
[-] ⛔ Change key ( failed )
[+] Setting ISODEP -> inactive
```
When I try to specify the same key for auth and newkey I get another error:

```
[usb] pm3 --> hf mfdes changekey --aid 111114 -n 0 -t des -k 1111111111111111 --newkeyno 0 --newalgo aes --newkey 11111111111111111111111111111111  -a -v
[=] Changing key for AID 111114
[=] auth key 0: des [8] 11 11 11 11 11 11 11 11 
[=] changing key number  0x00 (0)
[=] old key: des [8] 11 11 11 11 11 11 11 11 
[=] new key: aes [16] 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 11 
[=] new key version: 0x00
[=] Key num: 0 Key algo: des Key[8]: 11 11 11 11 11 11 11 11 
[=] Secure channel: n/a Command set: niso Communication mode: encrypt
[+] Setting ISODEP -> inactive
[+] >>>> 90 5A 00 00 03 14 11 11 00 
[+] Setting ISODEP -> inactive
[+] Setting ISODEP -> NFC-A
[+] <<<< 91 00 
[=] AID 111114 is selected
[=] Auth: cmd: 0x1a keynum: 0x00
[+] >>>> 90 1A 00 00 01 00 00 
[+] <<<< 7E 2B 1B 40 50 9C 35 B0 91 AF 
[+] >>>> 90 AF 00 00 10 1B 94 D3 8C 97 47 28 B9 8D 92 3B 17 92 94 A7 8E 00 
[+] <<<< 23 11 5A 65 69 10 78 A5 91 00 
[=] Session key : 01 02 03 04 CA 1D 63 59 
[=] Desfire  authenticated
[+] >>>> 90 C4 00 00 19 00 2E 43 BB 53 A9 67 48 02 8E 35 01 3C 72 A4 33 B7 F2 06 60 2D 62 A7 52 4F 00 
[+] <<<< 91 1E 
[!!] 🚨 APDU(90c4) ERROR: [0x911E] CRC or MAC does not match data
[-] ⛔ Change key ( failed )
[+] Setting ISODEP -> inactive

```

Am I doing something wrong?

iceman1001, 2025-06-08:
Use the discord server to get help from the community.

The DESFire functionality is not the same as NXP production level.   So for us its more a guessing game. 
You will need to look at the trace list.  `hf mfdes list`  to see the communications better.

OevreFlataeker, 2025-09-20:
I'll have a look! Thanks for pinging me!

OevreFlataeker, 2025-09-20:
Unfortunately the behavior is unchanged. I get exactly the same error when I use the latest sources

merlokk, 2025-09-22:
The key should be the same type. The type of the key is set by creating the application