RfidResearchGroup/proxmark3 #2878: Identifying hardware failuree

DeepLearn6984, 2025-06-03:
***Compilation problems***
None.

***flashing problems***
I flashed the previous stable and the last one. No issue during the flash process for bootloader and app.

```
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 65% used )

    Client.... Iceman/master/v4.18994-552-g325376d7e 2024-11-21 20:36:26
    Bootrom... Iceman/master/v4.18994-552-g325376d7e-suspect 2024-11-21 20:35:52
    OS........ Iceman/master/v4.18994-552-g325376d7e-suspect 2024-11-21 20:36:00
    Target.... PM3 GENERIC
```

**Describe the bug**
I'm working with a Proxmark3 Easy. The problem is that I am unable to read HF or LF tags. One month ago, it was working perfectly with the proxmark3-v4.20142 release, but now every operation is failing.

I tried reflashing the same release, and also went back to v4.18994, but that didn't change anything.

I was considering a hardware issue, but aside from the inability to read, everything else seems to be working correctly. I also tried loosening and tightening the antenna screws to check for a bad connection, but that didn’t help either.

**To Reproduce**
[usb] pm3 --> hf mf auto **(B led is getting RED)**
[usb] pm3 --> hf mf auto (B is RED)
[usb] pm3 --> hf mf auto (B is RED)
[!] Communicating with Proxmark3 device failed

[+] Communicating with PM3 over USB-CDC
[usb] pm3 -->

> Board seems to reset at third attemp. After it reboots, led B is cleared. Issue happen again each time I type: hf mf auto

**Expected behavior**
Tag is read without error.

**Screenshots**
```
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 65% used )

    Client.... Iceman/master/v4.18994-552-g325376d7e 2024-11-21 20:36:26
    Bootrom... Iceman/master/v4.18994-552-g325376d7e-suspect 2024-11-21 20:35:52
    OS........ Iceman/master/v4.18994-552-g325376d7e-suspect 2024-11-21 20:36:00
    Target.... PM3 GENERIC

[=] No previous history could be loaded
[usb] pm3 --> hf mf auto
[usb] pm3 --> hf mf auto
[usb] pm3 --> hf mf auto

[!] Communicating with Proxmark3 device failed

[offline] pm3 --> hf mf auto
```

**Desktop (please complete the following information):**
 - OS: w10
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
```
 [usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.18994-552-g325376d7e-suspect 2024-11-21 20:36:26 a3319015f
  compiled with............. MinGW-w64 13.2.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Python SWIG support....... absent
  Lua script support........ present ( 5.4.6 )
  Lua SWIG support.......... present

 [ Proxmark3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.18994-552-g325376d7e-suspect 2024-11-21 20:35:52 a3319015f
       os: Iceman/master/v4.18994-552-g325376d7e-suspect 2024-11-21 20:36:00 a3319015f
  compiled with GCC 12.2.0

 [ FPGA ]
 fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
 fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
 fpga_pm3_felica.ncd image 2s30vq100 2024-02-03 15:12:41
 fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 65% used )
```
 - hw status
```
 [usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 41196
[#]   Available memory........ 41196
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | n/a | n/a |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | n/a | n/a |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | n/a | n/a |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 300544
[#]   Transfer Speed PM3 -> Client... 601088 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 32181 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#]
```
 - hw tune
```
[=] -------- Reminder ----------------------------
[=] `hw tune` doesn't actively tune your antennas.
[=] It's only informative.
[=] Measuring antenna characteristics...
[/] 10

[=] -------- LF Antenna ----------
[+] 125.00 kHz ........... 23.62 V
[+] 134.83 kHz ........... 16.67 V
[+] 121.21 kHz optimal.... 24.41 V
[+]
[+] Approx. Q factor measurement
[+] Frequency bandwidth... 6.1
[+] Peak voltage.......... 7.1
[+] LF antenna............ ok

[=] -------- HF Antenna ----------
[+] 13.56 MHz............. 15.57 V
[+]
[+] Approx. Q factor measurement
[+] Peak voltage.......... 4.5
[+] HF antenna ( ok )

[=] -------- LF tuning graph ------------
[+] Orange line - divisor 95 / 125.00 kHz
[+] Blue line - divisor   88 / 134.83 kHz

[=] Q factor must be measured without tag on the antenna

```
**Additional context**
Add any other context about the problem here.

iceman1001, 2025-06-05:
Try pulling latest source, compile,  flash all,  and re-run your tests.

DeepLearn6984, 2025-06-05:
I just started a clean build from the git. No issue during the build. Flash boot and app without issue.

End of the build:
`[=] CXX brute_key
[-] CC mfd_aes_brute.c
[=] CXX mfd_aes_brute
[-] CC mfd_multi_brute.c
[=] CXX mfd_multi_brute
[*] MAKE cryptorf/all
[-] CC cm.c
[-] CC ../../common/cryptorf/cryptolib.c
[-] CC util.c
[=] CXX cm
[-] CC sm.c
[=] CXX sm
[-] CXX sma.cpp
[=] CXX sma
[-] CXX sma_multi.cpp
[=] CXX sma_multi
pm3 ~/proxmark3_git$ ls`

Both flash ok: 
```
pm3 ~/proxmark3_git$ pm3-flash-bootrom
[=] Session log C:\ProxSpace\pm3/.proxmark3/logs/log_20250605161350.txt
[+] loaded `C:\ProxSpace\pm3/.proxmark3/preferences.json`
[+] About to use the following file:
[+]    C:\ProxSpace\pm3\proxmark3_git\client\../bootrom/obj/bootrom.elf
[+] Loading ELF file C:\ProxSpace\pm3\proxmark3_git\client\../bootrom/obj/bootrom.elf
[+] ELF file version Iceman/master/v4.20142-232-ge2a1f30b4-dirty-suspect 2025-06-05 18:11:33 258eb9620

[+] Waiting for Proxmark3 to appear on COM5
[|] 59 found
[+] Entering bootloader...
[+] (Press and release the button only to abort)
[+] Trigger restart...
[+] Waiting for Proxmark3 to appear on COM5
[-] 59 found
[=] Available memory on this board: 512K bytes

[=] Permitted flash range: 0x00100000-0x00180000
[+] Loading usable ELF segments:
[+]    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1: V 0x00200000 P 0x00100200 (0x00000f2c->0x00000f2c) [R X] @0x298

[+] Flashing...
[+] Writing segments for file: C:\ProxSpace\pm3\proxmark3_git\client\../bootrom/obj/bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
. ok
[+]  0x00100200..0x0010112b [0xf2c / 8 blocks]
........ ok

[+] All done

[=] Have a nice day!

pm3 ~/proxmark3_git$ pm3-flash-fullimage
[=] Session log C:\ProxSpace\pm3/.proxmark3/logs/log_20250605161404.txt
[+] loaded `C:\ProxSpace\pm3/.proxmark3/preferences.json`
[+] About to use the following file:
[+]    C:\ProxSpace\pm3\proxmark3_git\client\../armsrc/obj/fullimage.elf
[+] Loading ELF file C:\ProxSpace\pm3\proxmark3_git\client\../armsrc/obj/fullimage.elf
[+] ELF file version Iceman/master/v4.20142-232-ge2a1f30b4-dirty-suspect 2025-06-05 18:12:04 258eb9620

[+] Waiting for Proxmark3 to appear on COM5
[|] 59 found
[+] Entering bootloader...
[+] (Press and release the button only to abort)
[+] Trigger restart...
[+] Waiting for Proxmark3 to appear on COM5
[-] 59 found
[=] Available memory on this board: 512K bytes

[=] Permitted flash range: 0x00102000-0x00180000
[+] Loading usable ELF segments:
[+]    0: V 0x00102000 P 0x00102000 (0x00053e7c->0x00053e7c) [R X] @0x98
[+]    1: V 0x00200000 P 0x00155e7c (0x00001d2e->0x00001d2e) [R X] @0x53f18
[=] Note: Extending previous segment from 0x53e7c to 0x55baa bytes

[+] Flashing...
[+] Writing segments for file: C:\ProxSpace\pm3\proxmark3_git\client\../armsrc/obj/fullimage.elf
[+]  0x00102000..0x00157ba9 [0x55baa / 686 blocks]
...................................................................
        @@@  @@@@@@@ @@@@@@@@ @@@@@@@@@@   @@@@@@  @@@  @@@
        @@! !@@      @@!      @@! @@! @@! @@!  @@@ @@!@!@@@
        !!@ !@!      @!!!:!   @!! !!@ @!@ @!@!@!@! @!@@!!@!
        !!: :!!      !!:      !!:     !!: !!:  !!! !!:  !!!
        :    :: :: : : :: :::  :      :    :   : : ::    :
        .    .. .. . . .. ...  .      .    .   . . ..    .
...................................................................
...................................................................
...................................................................
................................................. ok

[+] All done

[=] Have a nice day!
```

Then same comm issue:
```
pm3 ~/proxmark3_git$ pm3
[=] Session log C:\ProxSpace\pm3/.proxmark3/logs/log_20250605161433.txt
[+] loaded `C:\ProxSpace\pm3/.proxmark3/preferences.json`
[+] Using UART port COM5
[+] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.
  888   Y88b 8888b   d8888 d88P  Y88b
  888    888 88888b.d88888      .d88P
  888   d88P 888Y88888P888     8888"
  8888888P"  888 Y888P 888      "Y8b.
  888        888  Y8P  888 888    888
  888        888   "   888 Y88b  d88P
  888        888       888  "Y8888P"

  [ Dum spiro, spero!  ]

  [ Proxmark3 ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 69% used )
    Target.... PM3 GENERIC

    Client.... Iceman/master/v4.20142-232-ge2a1f30b4-dirty 2025-06-05 18:10:12
    Bootrom... Iceman/master/v4.20142-232-ge2a1f30b4-dirty-suspect 2025-06-05 18:11:33 258eb9620
    OS........ Iceman/master/v4.20142-232-ge2a1f30b4-dirty-suspect 2025-06-05 18:12:04 258eb9620

[usb] pm3 --> hf mf auto

[!] Communicating with Proxmark3 device failed

[+] Communicating with PM3 over USB-CDC
[usb] pm3 --> hf mf auto
[usb] pm3 --> hf mf auto

[!] Communicating with Proxmark3 device failed
[+] Communicating with PM3 over USB-CDC
[usb] pm3 --> hf mf info
[usb] pm3 --> hf mf info
[!] Failed to get current device debug level

[!] Communicating with Proxmark3 device failed

[+] Communicating with PM3 over USB-CDC
[usb] pm3 -->
```

iceman1001, 2025-06-05:
```
hw status
hw version
hw tune
```

DeepLearn6984, 2025-06-06:
hw status: 
```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 39968
[#]   Available memory........ 39968
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 05-06-2025 18:05:42
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | n/a | n/a |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | n/a | n/a |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | n/a | n/a |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override........... std    ( follow standard )
[#]   [b] BCC override............... std    ( follow standard )
[#]   [2] CL2 override............... std    ( follow standard )
[#]   [3] CL3 override............... std    ( follow standard )
[#]   [r] RATS override.............. std    ( follow standard )
[#]   [m] Magsafe polling............ disabled
[#]   [p] Polling loop annotation.... disabled 00000000000000000000000000000000
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 294912
[#]   Transfer Speed PM3 -> Client... 589824 bytes/s
[#] Various
[#]   Max stack usage..... 3520 / 8480 bytes
[#]   Debug log level..... 1 ( error )
[#]   ToSendMax........... -1
[#]   ToSend BUFFERSIZE... 2308
[#]   Slow clock.......... 32168 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#]
[usb] pm3 -->
```

hw version:
```
[usb] pm3 --> hw version

 [ Proxmark3 ]

 [ Client ]
  Iceman/master/v4.20142-232-ge2a1f30b4-dirty-suspect 2025-06-05 18:10:12 258eb9620
  Compiler.................. MinGW-w64 13.2.0
  Platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  Native BT support......... absent
  Python script support..... present ( 3.11.5 )
  Python SWIG support....... present
  Lua script support........ present ( 5.4.6 )
  Lua SWIG support.......... present

 [ Model ]
  Firmware.................. PM3 GENERIC

 [ ARM ]
  Bootrom.... Iceman/master/v4.20142-232-ge2a1f30b4-dirty-suspect 2025-06-05 18:11:33 258eb9620
  OS......... Iceman/master/v4.20142-232-ge2a1f30b4-dirty-suspect 2025-06-05 18:12:04 258eb9620
  Compiler... GCC 12.2.0

 [ FPGA ]
 fpga_pm3_hf.ncd image 2s30vq100 05-06-2025 18:05:42
 fpga_pm3_lf.ncd image 2s30vq100 05-06-2025 18:05:42
 fpga_pm3_felica.ncd image 2s30vq100 05-06-2025 18:05:42
 fpga_pm3_hf_15.ncd image 2s30vq100 05-06-2025 18:05:42

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 69% used )

[usb] pm3 -->
```

hw tune:
```
[usb] pm3 --> hw tune

[=] -------- Reminder ----------------------------
[=] `hw tune` doesn't actively tune your antennas.
[=] It's only informative.
[=] Measuring antenna characteristics...
[|] 10

[=] -------- LF Antenna ----------
[+] 125.00 kHz ........... 23.80 V
[+] 134.83 kHz ........... 16.82 V
[+] 122.45 kHz optimal.... 24.60 V
[+]
[+] Approx. Q factor measurement
[+] Frequency bandwidth... 6.4
[+] Peak voltage.......... 7.2
[+] LF antenna............ ok

[=] -------- HF Antenna ----------
[+] 13.56 MHz............. 15.57 V
[+]
[+] Approx. Q factor measurement
[+] Peak voltage.......... 4.5
[+] HF antenna ( ok )

[=] -------- LF tuning graph ------------
[+] Orange line - divisor 95 / 125.00 kHz
[+] Blue line - divisor   88 / 134.83 kHz

[=] Q factor must be measured without tag on the antenna

[usb] pm3 -->
```

iceman1001, 2025-06-06:
nothing wrong there,   which tag are you trying to read?

DeepLearn6984, 2025-06-06:
I'm trying to read mifare tags that used to work properly before.

```
[usb] pm3 --> hf search
[|] Searching for ThinFilm tag...[!] timeout while waiting for reply
[-] Searching for ISO14443-A tag...[!] Failed to get current device debug level
[|] Searching for Fuji/Xerox tag...
[!] Communicating with Proxmark3 device failed

[-] Searching for ISO14443-B tag...
```

iceman1001, 2025-06-06:
try  `hf 14a reader -@"   and see when you get good UID reads
have you fiddled with the antenna?  like unscrewed it?   If so, make sure its correct reassembled.

DeepLearn6984, 2025-06-06:
I ran the command but still issue while communicating. Otherwise I screwed the antenna tightly as it was originaly.

```
[usb] pm3 --> hf 14a reader -@
[=] Press <Enter> to exit
[=] Waiting for a response from the Proxmark3...
[=] You can cancel this operation by pressing the pm3 button
[-] Timed out while trying to download data from device
[!] timeout while waiting for reply
[=] Waiting for a response from the Proxmark3...
[=] You can cancel this operation by pressing the pm3 button
[-] Timed out while trying to download data from device
[!] timeout while waiting for reply

[!] Communicating with Proxmark3 device failed

[=] Waiting for a response from the Proxmark3...
[=] You can cancel this operation by pressing the pm3 button
[-] Timed out while trying to download data from device
[!] timeout while waiting for reply
```

iceman1001, 2025-06-06:
It's nothing wrong with the software/firmware as far as I can tell. 
I have run latest source on a RDV4 today and I can't replicate your issue.  

Its not an issue for this repo since its related to software/firmware. 

Use the discord server and see if the community can help you out.

closed 2025-06-06 by iceman1001