RfidResearchGroup/proxmark3 #2908: Darkside Is not working anymore

DREAMSOLDIERc404, 2025-06-16:
***Photo of the problem***

[image]

***Explaining the problem***

I tried with 4 different tags from different manufacturers, two legit:
- A water dispenser
- An access badge  

Two Chinese:
- mf-UID
- Magic gen2

The same problem appeared on all of the! Finally I add that they are not static nonce tags.

It is really annoying, I tried all 4 with mfcuk (both 3.8 and 3.3/rev65) and it did not give any error signs.
At this point it is not a strange behavior but a code error, I ask for your collaboration to reach a solution together.
Does it give you the same error or have you had success in darkside attacks with the latest updates?

iceman1001, 2025-06-17:
what does `hf mf info` say about all your four cards?

iceman1001, 2025-06-17:
Are you on the latest source?

And I will take a look at some MFC cards later today

iceman1001, 2025-06-17:
the question now is darkside stopped working.   You can do a `git bisect`  and find the offending change

DREAMSOLDIERc404, 2025-06-17:
i was searching for a good commit but i got stuck in a compatibility exception with strings format due to some package too old version required, but i managed to see that 

    commit 7368cccdf90b4c68ae598ff27ee2e9db5b71e1a4 (HEAD -> master)
    Author: iceman1001 <iceman@iuse.se>
    Date:   Mon May 29 06:57:39 2023 +0200
    
    Disney MagicBand

is still a bad commit and it's very old so i don't know if i can go to search for older commit or assume that it could be 2 other possibily reasons:

1. Problems of compatibilty with my reader (or wrong configuraton file setted)
2.  The tags are the problem but i can't find why
3. + It could be another softrware problem not considered

DREAMSOLDIERc404, 2025-06-17:
Yes, I agree with you... but you can do a test on your proxmark?
Oh right because i have a pm3 generic, but before you say that is chinese garbage i want to ask if you could run some rapid tests pleaseeeee 😢

DREAMSOLDIERc404, 2025-06-25:
I'd like a response... If there's something else i can do, can you tell me please? 🥺

iceman1001, 2025-06-29:
Not much to say about it,  it doesn't work currently.   My time is limited so I will leave it here open for others to help