RfidResearchGroup/proxmark3 #2943: Segmentation Fault When Sniffing DESFire EM Card (hf 14a sniff -c)

boby498800, 2025-07-22:
Hi Iceman and team,

I'm encountering a segmentation fault when attempting to sniff a DESFire EV1 emulated card using the command hf 14a sniff -c.

Here's the context:

[+] ------------------------------------ PICC level -------------------------------------
[+] # applications....... 1
[+] 
[+] PICC level auth commands
[+]    Auth.............. NO
[+]    Auth ISO.......... NO
[+]    Auth AES.......... YES
[+]    Auth Ev2.......... YES
[+]    Auth ISO Native... YES
[+]    Auth LRP.......... NO
[+] PICC level rights
[+] [1...] CMK Configuration changeable               : YES
[+] [.1..] CMK required for create/delete             : NO
[+] [..1.] CMK required for AID list / GetKeySettings : NO
[+] [...1] CMK is changeable                          : YES
[+] 
[+] Key type... AES
[+] Key cnt.... 1
[+] PICC key 0 version: 0 (0x00)
 [=] --- Free memory
[+]    Available free memory on card... 4544 bytes

When I run hf 14a sniff -c, the command fails with:

[=] Standalone DESFire
[usb] pm3 --> Hf 14a sniff -c
[=] Press pm3 button to abort sniffing
[usb] pm3 --> /usr/local/bin/pm3: line 253: 3626 Segmentation fault: 11  $CLIENT "$@"

System Info:

Device: Proxmark3 RDV4

MCU: AT91SAM7S512 Rev A (512 KB, 76% used)

OS: macOS Sonoma 14.7

Client: Iceman/master/v4.20469 (built 2025-06-16)

Bootrom: Iceman/master/v4.20469-suspect (72b1b17a3)

OS: Iceman/master/v4.20469-suspect (72b1b17a3)

Any ideas on what could be causing this segfault or how to debug further?

Thanks for all your work on this awesome tool!

iceman1001, 2025-07-22:
A known issue which should have been fixed in latest source. 

Pull latest source and try again.

boby498800, 2025-07-22:
Solved with brew install --HEAD --with-blueshark proxmark3 for blueshark support, latest non-stable from GitHub .

Thank you Iceman