RfidResearchGroup/proxmark3 #2945: Unknown card with backdoor, please report details!

Cyl0nius, 2025-07-25:
***Things to try before submitting bug report***
read the [troubleshooting guide](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to Iceman repo.

-

**Describe the bug**
A clear and concise description of what the bug is.
Please include text output of the bug happening.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - hw tune

**Additional context**
Add any other context about the problem here.

iceman1001, 2025-07-25:
Interesting but you seem to be on an older firmware/code base. 
If you update to latest source and run again.

iceman1001, 2025-09-03:
ping