RfidResearchGroup/proxmark3 #2954: Issues with Bluetooth on proxmark3

boby498800, 2025-08-11:
Hi everyone,
I’m trying to install blueshark on my Mac that’s partitioned with Ubuntu. I’ve followed the installation procedure, but I’m experiencing bluetooth connection drops/disconnects during the process.
Has anyone encountered a similar issue or have any ideas on how to resolve this? I’d really appreciate any suggestions or troubleshooting tips you might have.
Thanks in advance for your time and support!

[image]
[image]
https://github.com/user-attachments/assets/f0ea4398-75cf-4808-b147-5a2ef5c93b74

boby498800, 2025-08-14:
ISSUE FIXED : 
In the Malefile.platform file:
PLATFORM_EXTRA= BTADDON FPC_USART_DEV
Instead of just PLATFORM_EXTRA= BTADDON
Thank you.

closed 2025-08-14 by boby498800

iceman1001, 2025-08-14:
flag: `fpc_usart_dev`  shouldn't impact the connectivity being dropped randomly.     It enables two commands (If I remember correct) and being able to talk at the usart port from the pm3 device by issuing command over usb from the client,  to be executed against the usart interface on device.

I feel this issue is missing details.

boby498800, 2025-08-14:
[image]
[image]