RfidResearchGroup/proxmark3 #2955: Write timeout while changing emulator memory during iclass simulate

derickmachaa, 2025-08-12:
**The proxmark3 easy hangs during a brute force/simulation attempt on hf iclass simulation/bruteforce**

[usb] pm3 --> hf iclass encode -w H10301 --fc 32 --cn 1337 --emu
[+] Loaded 16 bytes from binary file `iclass_decryptionkey.bin`
[=] Uploading to emulator memory
[=] ..
[+] uploaded 32 bytes to emulator memory
[?] Hint: You are now ready to simulate. See `hf iclass sim -h`
[usb] pm3 --> hf iclass encode -w H10301 --fc 33 --cn 1337 --emu
[+] Loaded 16 bytes from binary file `iclass_decryptionkey.bin`
[=] Uploading to emulator memory
[=] ..
[+] uploaded 32 bytes to emulator memory
[?] Hint: You are now ready to simulate. See `hf iclass sim -h`
[usb] pm3 --> [!!] 🚨 UART:: write time-out

**To Reproduce**
1. Load a HID legacy iclass dump to emulator memory 'hf iclass eload -f test.bin'
2. start full simulation of the tag 'hf iclass sim -t 3'
3. start modifying the fc and cn numbers ' hf iclass encode -w H10301 --fc 150 --cn 150 --emu'
4. Memory is updated successfully
5. The first three attempts work great but it fails on the fourth attempt with UART:: write time-out

**Expected behavior**
The memory to updates to work seamlessly

 -  hw version
 [ Proxmark3 ]                  
                                               
 [ Client ]
  Iceman/master/v4.20469-180-g8ab2f2b5a-suspect 2025-08-12 23:47:24 ef5b2e843                  
  Compiler.................. GCC 15.1.1 20250425               
  Platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present                                                           
  Native BT support......... present                                                           
  Python script support..... present ( 3.13.5 )
  Python SWIG support....... present
  Lua script support........ present ( 5.4.8 )
  Lua SWIG support.......... present                                                           
                                               
 [ Model ]                   
  Firmware.................. PM3 GENERIC                                                       
                                                                                               
 [ ARM ]                        
  Bootrom.... Iceman/master/v4.20469-180-g8ab2f2b5a-suspect 2025-08-12 23:47:26 ef5b2e843      
  OS......... Iceman/master/v4.20469-180-g8ab2f2b5a-suspect 2025-08-12 23:47:41 ef5b2e843      
  Compiler... GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                                               
 [ FPGA ]                    
 fpga_pm3_hf.ncd image 2s30vq100 04-07-2025 16:24:32                                           
 fpga_pm3_lf.ncd image 2s30vq100 04-07-2025 16:24:32                                           
 fpga_pm3_felica.ncd image 2s30vq100 04-07-2025 16:24:32                                                                                                                                      
 fpga_pm3_hf_15.ncd image 2s30vq100 04-                                                        

 [ Hardware ]                   
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI       
  --= Internal SRAM size: 64K bytes                                                            
  --= Architecture identifier: AT91SAM7Sxx Series                                              
  --= Embedded flash memory 512K bytes ( 69% used )      

 [usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 40268
[#]   Available memory........ 36172
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf_15.ncd image 2s30vq100 04-07-2025 16:24:32
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | n/a | n/a |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | n/a | n/a |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | n/a | n/a |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override........... std    ( follow standard )
[#]   [b] BCC override............... std    ( follow standard )
[#]   [2] CL2 override............... std    ( follow standard )
[#]   [3] CL3 override............... std    ( follow standard )
[#]   [r] RATS override.............. std    ( follow standard )
[#]   [m] Magsafe polling............ disabled
[#]   [p] Polling loop annotation.... disabled 00000000000000000000000000000000
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 364032
[#]   Transfer Speed PM3 -> Client... 728064 bytes/s
[#] Various
[#]   Max stack usage..... 4024 / 8480 bytes
[#]   Debug log level..... 1 ( error )
[#]   ToSendMax........... 22
[#]   ToSend BUFFERSIZE... 2308
[#]   Slow clock.......... 30443 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)

[usb] pm3 --> hw tune
[=] -------- Reminder ----------------------------
[=] `hw tune` doesn't actively tune your antennas.
[=] It's only informative.
[=] Measuring antenna characteristics...
 🕛   9

[=] -------- LF Antenna ----------
[+] 125.00 kHz ........... 23.27 V
[+] 134.83 kHz ........... 15.98 V
[+] 120.00 kHz optimal.... 24.89 V
[+] 
[+] Approx. Q factor measurement
[+] Frequency bandwidth... 6.0
[+] Peak voltage.......... 7.2
[+] LF antenna............ ok

[=] -------- HF Antenna ----------
[+] 13.56 MHz............. 16.25 V
[+] 
[+] Approx. Q factor measurement
[+] Peak voltage.......... 4.7
[+] HF antenna ( ok )

[=] -------- LF tuning graph ------------
[+] Orange line - divisor 95 / 125.00 kHz
[+] Blue line - divisor   88 / 134.83 kHz

[=] Q factor must be measured without tag on the antenna

iceman1001, 2025-08-14:
Once in simulation mode,  the device doesn't "take" more commands even if the client is free to execute commands.
This is for doing client side stuff.  I never heard of someone sending updates to the emulator memory during but hey,  there is a first for everything.   

The correct procedure is that you will need to end the simulation with pressing the pm3 button first. 

For other simulations we have "exit" on receiving usb package during simulation.  iclass sim doesn't most likely for simulation is time sensitive.

closed 2025-08-14 by derickmachaa

iceman1001, 2025-08-14:
Sounds like an AI....

derickmachaa, 2025-08-14:
am not an AI, actually for me to speculate that emulator memory could be changed  while iclass simulation was active I did the following
1. loaded a full dump to memory with a known fc and cn
2. started simulation -t 3
3. hf iclass encoded to emulator memory with a different fc and cn :- no errors reported
4. tried to eview the memory but couldn't, no response
5. stopped simulation via keypress
6. did an eview of the emulator memory and it had the updated fc and cn

derickmachaa, 2025-08-14:
[image]

You can see this from the screenshot below, I just needed a quick and easy iclass brute