RfidResearchGroup/proxmark3 #2961: (Desfire) hf mfdes auth, createfile, and getfileids commands quit working with 2025-08-28 updates

trigat, 2025-08-29:
I noticed I receive the same error when trying to execute hf mfdes auth, createfile, and getfileids commands on the following version:

    Client.... Iceman/master/v4.20469-202-g599aa2243 2025-08-28 09:07:44                                                                                                                           
    Bootrom... Iceman/master/v4.20469-192-g6ee974b93-suspect 2025-08-21 09:40:35 9e9e48af6                                                                                                         
    OS........ Iceman/master/v4.20469-192-g6ee974b93-suspect 2025-08-21 09:40:45 9e9e48af6

I rolled back to the the this version and commands work as expected:

    Client.... Iceman/master/v4.20469 2025-06-16 16:18:01
    Bootrom... Iceman/master/v4.20469-suspect 2025-06-16 16:18:01 72b1b17a3
    OS........ Iceman/master/v4.20469-suspect 2025-06-16 16:18:01 72b1b17a3

I've tested this with DES, 3DES, AES, multiple AID numbers, and set DF Names.

Here is an example of the error I'm receiving:

`
hf mfdes createapp -t aes -k 00000000000000000000000000000000 --aid 111111 --fid 1111 --dstalgo aes -m plain -a

[=] Received MAC OK
[+] Desfire application 111111 successfully created

hf mfdes createfile --aid 111111 --fid 01 --isofid 0001 --size 001000 -t aes -k 00000000000000000000000000000000 -m plain -a -v

=] Desfire  authenticated
[=] App: 111111. File num: 0x01 type: 0x00 data[9]: 01 11 11 00 EE EE 00 10 00 
[=] ---- Create file settings ----
[+] File type        : Standard data
[+] File number      : 0x01 (1)
[+] File ISO number  : 0x1111
[+] File comm mode   : Plain
[+] Additional access: No
[+] Access rights    : EEEE
[+]   read......... free
[+]   write........ free
[+]   read/write... free
[+]   change....... free
[=] File size (bytes)... 4096 / 0x1000
[+] >>>> 90 CD 00 00 09 01 11 11 00 EE EE 00 10 00 00 
[+] <<<< 24 85 4C C4 C9 6E F2 31 91 00 
[=] Received MAC OK
[+] Standard data file 01 in the app 111111 created successfully

I can successfully write to the file, but I can't read the file or getfileids:

hf mfdes read --aid 111111 --fid 01 -t aes -k 00000000000000000000000000000000 -m plain -v -a

!!] 🚨 APDU(00a4) ERROR: [0x6A87] Lc inconsistent with P1-P2
[!!] 🚨 Desfire DF name select error
[-] ⛔ Select or authentication AID 111111 failed. Result [200] Can't select application.

hf mfdes getfileids --aid 111111 -t aes -k 00000000000000000000000000000000 -m plain -a -v

[!!] 🚨 APDU(00a4) ERROR: [0x6A87] Lc inconsistent with P1-P2
[!!] 🚨 Desfire DF name select error
[-] ⛔ Select or authentication AID 111111 failed. Result [200] Can't select application.
`

iceman1001, 2025-08-29:
I reverted some changes to bigbuf,   try pulling lastest / compile / flash,   and try again.

nvx, 2025-08-29:
I just tried with the latest commit (f0ea0d251aef24fa897ca3611f301928ed958272) using a DESFire EV2 card with those steps and it worked.

Can you confirm if it's fixed for you too now @trigat and if not what version of DESFire card are you using? (`hf mfd info` output would be handy)

trigat, 2025-08-29:
It looks like I'm still having some issues with auth, getfileids, freemem, etc...

USING:
        Client.... Iceman/master/v4.20469-209-gd6aeda707 2025-08-29 08:12:18
        Bootrom... Iceman/master/v4.20469-209-gd6aeda707-suspect 2025-08-29 08:12:03 2689d7032
        OS........ Iceman/master/v4.20469-209-gd6aeda707-suspect 2025-08-29 08:12:47 2689d7032
    
    [usb] pm3 --> hf mfdes formatpicc -t des -k 0000000000000000 -m plain -a -v
    [=] Key num: 0 Key algo: des Key[8]: 00 00 00 00 00 00 00 00
    [=] Secure channel: n/a Command set: niso Communication mode: plain
    [+] Setting ISODEP -> inactive
    [+] Setting ISODEP -> NFC-A
    [=] Anticollision ok
    [=] Auth: cmd: 0x1a keynum: 0x00
    [+] >>>> 90 1A 00 00 01 00 00
    [+] <<<< 83 B8 79 40 92 32 72 59 91 AF
    [+] >>>> 90 AF 00 00 10 CC 45 06 66 5C 22 9B 49 BB D7 B6 71 13 A9 42 F3 00
    [+] <<<< 03 FB 11 80 0B 82 04 95 91 00
    [=] Session key : 01 02 03 04 99 D2 F4 FF
    [=] Desfire  authenticated
    [+] >>>> 90 FC 00 00 00
    [+] <<<< 0A 28 19 F8 0B A0 56 A0 91 00
    [=] Received MAC OK
    [+] Desfire format: done!
    [+] Setting ISODEP -> inactive
    
    [usb] pm3 --> hf mfdes createapp -t des -k 0000000000000000 --aid 111111 --fid 1111 --dstalgo des -m plain -a
    [+] Setting ISODEP -> inactive
    [+] Setting ISODEP -> NFC-A
    [+] >>>> 90 1A 00 00 01 00 00
    [+] <<<< DA 99 62 AB E4 4F AF BA 91 AF
    [+] >>>> 90 AF 00 00 10 A2 7E 07 8F 00 DF B2 E6 A5 59 A6 37 08 B1 30 17 00
    [+] <<<< D6 11 BA B5 9D ED E6 75 91 00
    [+] >>>> 90 CA 00 00 07 11 11 11 0F 2E 11 11 00
    [+] <<<< A4 E6 7B D7 FF 6B 41 6B 91 00
    [=] Received MAC OK
    [+] Desfire application 111111 successfully created
    [+] Setting ISODEP -> inactive
    
    [usb] pm3 --> hf mfdes info
    
    [=] ---------------------------------- Tag Information ----------------------------------
    [+]               UID: 04 58 43 82 28 19 90
    [+]      Batch number: 20 76 65 30 30
    [+]   Production date: week 28 / 2023
    [+]      Product type: MIFARE DESFire native IC (physical card)
    
    [=] --- Hardware Information
    [=]    raw: 04010133001A05
    [=]      Vendor Id: NXP Semiconductors Germany
    [=]           Type: 0x01 ( DESFire )
    [=]        Subtype: 0x01
    [=]        Version: 33.0 ( DESFire EV3 )
    [=]   Storage size: 0x1A ( 8192 bytes )
    [=]       Protocol: 0x05 ( ISO 14443-2, 14443-3 )
    
    [=] --- Software Information
    [=]    raw: 04010103001A05
    [=]      Vendor Id: NXP Semiconductors Germany
    [=]           Type: 0x01 ( DESFire )
    [=]        Subtype: 0x01
    [=]        Version: 3.0
    [=]   Storage size: 0x1A ( 8192 bytes )
    [=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )
    
    [=] --------------------------------- Card capabilities ---------------------------------
    [=] 	3.0 - DESFire Ev3, Originality check, proximity check, badass EAL6
    
    [=] --- Tag Signature
    
[... 106 lines cut ...]
    [+] Setting ISODEP -> inactive
    [+] Setting ISODEP -> NFC-A
    [+] <<<< 91 00 
    [=] AID 111111 is selected
    [=] Auth: cmd: 0x1a keynum: 0x00
    [+] >>>> 90 1A 00 00 01 00 00 
    [+] <<<< 1A ED 31 78 4F C0 BF 44 91 AF 
    [+] >>>> 90 AF 00 00 10 DD BA B6 84 00 1D BA B0 96 B3 43 3F 76 3E 53 5A 00 
    [+] <<<< A4 4F 20 F8 76 8C 9A A3 91 00 
    [=] Session key : 01 02 03 04 A7 A7 BD 4F 
    [=] Desfire  authenticated
    [+] Application AID 111111 selected and authenticated succesfully
    [+] Context: 
    [=] Key num: 0 Key algo: des Key[8]: 00 00 00 00 00 00 00 00 
    [=] Secure channel: ev1 Command set: niso Communication mode: plain
    [=] Session key [8]: 01 02 03 04 A7 A7 BD 4F  
    [=]     IV [8]: 00 00 00 00 00 00 00 00 
    [+] Setting ISODEP -> inactive
    
    [usb] pm3 --> hf mfdes getfileids --aid 111111 -t des -k 0000000000000000 -m plain
    [=] ---- File ID list ----
    [=] File ID: 01
    
    [usb] pm3 --> hf mfdes freemem
    [+] Free memory [0x000e60] 3680 bytes

iceman1001, 2025-08-29:
@trigat  I pushed a test.   `hf mfdes default`  now always "reset"   selectdfname array and length when called. 

try calling it and see if your other commands work afterwards

trigat, 2025-08-29:
Hopefully I'm testing this correctly...

        Client.... Iceman/master/v4.20469-210-g1a626bc4f 2025-08-29 12:02:39
        Bootrom... Iceman/master/v4.20469-210-g1a626bc4f-suspect 2025-08-29 12:02:27 2689d7032
        OS........ Iceman/master/v4.20469-210-g1a626bc4f-suspect 2025-08-29 12:03:06 2689d7032
    
    [usb] pm3 --> hf mfdes default
    [=] -----------Default parameters---------------------------------
    [=] Key Num     : 0
    [=] Algo        : 2tdea
    [=] Key         : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
    [=] KDF algo    : none
    [=] KDF input   : [0] 
    [=] Secure chan : ev1
    [=] Command set : niso
    [=] Comm mode   : plain
    [usb] pm3 --> hf mfdes auth --aid 111111 -a -v
    
    [usb] pm3 --> hf mfdes info
    [+]               UID: 04 58 43 82 28 19 90
    [+] --- AID list ( 1 found )
    [+] 111111, 
    [+] Key type... 2TDEA
    [+] Key cnt.... 1
    
    [usb] pm3 --> hf mfdes chk -f mfdes_default_keys
    [!!] 🚨 Checking aid 0x111111...
    [=] Check: DES 2TDEA   keys: 00 
    [+] AID 0x111111, Found DES Key 00... 00 00 00 00 00 00 00 00 
    [+] AID 0x111111, Found 2TDEA Key 00... 00000000000000000000000000000000
    
    [usb] pm3 --> hf mfdes auth --aid 111111 -t 2TDEA -k 00000000000000000000000000000000 -v -a
    [!!] 🚨 APDU(00a4) ERROR: [0x6A87] Lc inconsistent with P1-P2
    [!!] 🚨 Desfire DF name select error
    [+] Setting ISODEP -> inactive
    [-] ⛔ Select or authentication AID 111111 failed. Result [200] Can't select application.
    
    [usb] pm3 --> hf mfdes auth --aid 111111
    [!!] 🚨 Desfire DF name select error
    [-] ⛔ Select or authentication AID 111111 failed. Result [200] Can't select application.
    
    [usb] pm3 --> hf mfdes freemem -t 2TDEA -k 00000000000000000000000000000000 -v -a
    [=] Key num: 0 Key algo: 2tdea Key[16]: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
    [=] Secure channel: n/a Command set: niso Communication mode: mac
    [+] Setting ISODEP -> inactive
    [+] >>>> 00 A4 04 00 03 00 00 00 00
    [!!] 🚨 APDU(00a4) ERROR: [0x6A82] File not found
    [!!] 🚨 Desfire DF name select error
    
    [usb] pm3 --> hf mfdes freemem
    [!!] 🚨 Desfire DF name select error
    
    [usb] pm3 --> hf mfdes getfileids --aid 111111 -t 2TDEA -k 00000000000000000000000000000000 -m plain -a -v
    [!!] 🚨 APDU(00a4) ERROR: [0x6A87] Lc inconsistent with P1-P2
    [!!] 🚨 Desfire DF name select error
    
    [usb] pm3 --> hf mfdes getfileids --aid 111111
    [!!] 🚨 Desfire DF name select error
    [-] ⛔ Select or authentication AID 111111 failed. Result [200] Can't select application.

kormax, 2025-08-30:
If you're using your forked repository in git, verify that you've synced it with the original one before pulling and building.

The fork present in your account wasn't updated for 5 days, and it would explain why the issue didn't go away after the changes.

trigat, 2025-08-30:
I've been pulling directly from https://github.com/RfidResearchGroup/proxmark3.

trigat, 2025-08-31:
Ahhh, good to know.
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Client platform:   Linux
compiler version:  cc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
x86_64
arm-none-eabi-gcc (15:13.2.rel1-2) 13.2.1 20231009