RfidResearchGroup/proxmark3 #2978: During sniff, when pressing PM3 button it crashes with Segmentation fault: 11  $CLIENT "$@"

yackermann, 2025-09-10:
**Describe the bug**
A clear and concise description of what the bug is.
Please include text output of the bug happening.

```
➜  ~ pm3
[=] Session log /Users/werk/.proxmark3/logs/log_20250910220026.txt
[+] loaded `/Users/werk/.proxmark3/preferences.json`
[+] Using UART port /dev/tty.usbmodem2101
[+] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.
  888   Y88b 8888b   d8888 d88P  Y88b
  888    888 88888b.d88888      .d88P
  888   d88P 888Y88888P888     8888"
  8888888P"  888 Y888P 888      "Y8b.
  888        888  Y8P  888 888    888
  888        888   "   888 Y88b  d88P
  888        888       888  "Y8888P"

Release v4.20469 - Daddy Iceman
  [ too many secrets! ☕ ]

  [ Proxmark3 ]

    MCU....... AT91SAM7S512 Rev B
    Memory.... 512 KB ( 77% used )
    Target.... RDV4

    Client.... Iceman/master/v4.20469 2025-06-16 16:18:01
    Bootrom... Iceman/master/v4.20469-suspect 2025-06-16 16:18:01 72b1b17a3
    OS........ Iceman/master/v4.20469-suspect 2025-06-16 16:18:01 72b1b17a3

[usb] pm3 --> hf 14a sniff -c -r
[=] Press pm3 button to abort sniffing
[usb] pm3 -->
[usb] pm3 -->
[usb] pm3 --> /opt/homebrew/bin/pm3: line 253: 51474 Segmentation fault: 11  $CLIENT "$@"
```

**To Reproduce**
Steps to reproduce the behavior:
1. [usb] pm3 --> hf 14a sniff -c -r
2. Do the NFC, then press the button
3. [usb] pm3 --> /opt/homebrew/bin/pm3: line 253: 51474 Segmentation fault: 11  $CLIENT "$@"

**Expected behavior**
Not crashing *)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  MacOS 15.6.1
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version 
```
 [ Proxmark3 ]

 [ Client ]
  Iceman/master/v4.20469-suspect 2025-06-16 16:18:01 72b1b17a3
  Compiler.................. Clang/LLVM Apple LLVM 17.0.0 (clang-1700.0.13.5)
  Platform.................. OSX / aarch64
  Readline support.......... present
  QT GUI support............ absent
  Native BT support......... absent
  Python script support..... present ( 3.13.7 )
  Python SWIG support....... present
  Lua script support........ present ( 5.4.8 )
  Lua SWIG support.......... present

 [ Model ]
  Device.................... RDV4
  Firmware.................. RDV4
  External flash............ present
  Smartcard reader.......... absent
  FPC USART for BT add-on... absent

 [ ARM ]
  Bootrom.... Iceman/master/v4.20469-suspect 2025-06-16 16:18:01 72b1b17a3
  OS......... Iceman/master/v4.20469-suspect 2025-06-16 16:18:01 72b1b17a3
  Compiler... GCC 13.3.1 20240614

 [ FPGA ]
 fpga_pm3_hf.ncd image 2s30vq100 16-06-2025 14:18:01
 fpga_pm3_lf.ncd image 2s30vq100 16-06-2025 14:18:01
 fpga_pm3_felica.ncd image 2s30vq100 16-06-2025 14:18:01
 fpga_pm3_hf_15.ncd image 2s30vq100 16-06-2025 14:18:01

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 77% used )
```

 - hw status

```
[usb] pm3 --> hw version

 [ Proxmark3 ]

 [ Client ]
  Iceman/master/v4.20469-suspect 2025-06-16 16:18:01 72b1b17a3
  Compiler.................. Clang/LLVM Apple LLVM 17.0.0 (clang-1700.0.13.5)
  Platform.................. OSX / aarch64
  Readline support.......... present
  QT GUI support............ absent
  Native BT support......... absent
  Python script support..... present ( 3.13.7 )
  Python SWIG support....... present
  Lua script support........ present ( 5.4.8 )
  Lua SWIG support.......... present

 [ Model ]
  Device.................... RDV4
  Firmware.................. RDV4
  External flash............ present
  Smartcard reader.......... absent
  FPC USART for BT add-on... absent

 [ ARM ]
  Bootrom.... Iceman/master/v4.20469-suspect 2025-06-16 16:18:01 72b1b17a3
  OS......... Iceman/master/v4.20469-suspect 2025-06-16 16:18:01 72b1b17a3
  Compiler... GCC 13.3.1 20240614

 [ FPGA ]
 fpga_pm3_hf.ncd image 2s30vq100 16-06-2025 14:18:01
 fpga_pm3_lf.ncd image 2s30vq100 16-06-2025 14:18:01
 fpga_pm3_felica.ncd image 2s30vq100 16-06-2025 14:18:01
 fpga_pm3_hf_15.ncd image 2s30vq100 16-06-2025 14:18:01

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 77% used )

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 37964
[#]   Available memory........ 37964
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 1519
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 16-06-2025 14:18:01
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Mfr ID / Dev ID......... EF / 11
[#]   JEDEC Mfr ID / Dev ID... EF / 3012
[#]   Memory size............. 256 kB (4 pages * 64k)
[#]   Unique ID (be).......... 0x272B6323525464D5
[#] Smart card module (ISO 7816)
[#]   version................. ( fail )
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | n/a | n/a |
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | n/a | n/a |
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | n/a | n/a |
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
[#]
[#] HF 14a config
[#]   [a] Anticol override........... std    ( follow standard )
[#]   [b] BCC override............... std    ( follow standard )
[#]   [2] CL2 override............... std    ( follow standard )
[#]   [3] CL3 override............... std    ( follow standard )
[#]   [r] RATS override.............. std    ( follow standard )
[#]   [m] Magsafe polling............ disabled
[#]   [p] Polling loop annotation.... disabled 00000000000000000000000000000000
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 378880
[#]   Transfer Speed PM3 -> Client... 757760 bytes/s
[#] Various
[#]   Max stack usage..... 3520 / 8480 bytes
[#]   Debug log level..... 1 ( error )
[#]   ToSendMax........... -1
[#]   ToSend BUFFERSIZE... 2308
[#]   Slow clock.......... 30277 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   Mifare... 0 keys - dict_mf.bin
[#]   T55xx.... 0 keys - dict_t55xx.bin
[#]   iClass... 0 keys - dict_iclass.bin
[#]
```
 - hw tune

**Additional context**
Add any other context about the problem here.

iceman1001, 2026-09-12:
ping

iceman1001, 2026-09-12:
Could be a stack issue on MacOs.    Anyway,  closing since no response from OP

closed 2026-09-12 by iceman1001